LordOfMice/hidusbf #1: Danger Label?

swineadam, 2019-01-14:
hello. i might have missed it, but... it doesn't appear that you warn about the danger of setting a higher poll rate to use with an optical mouse that otherwise wouldn't support it. doing this may eventually lead to the laser burning out and no longer working.

i have actually had this happen long ago in the past on two different computer mouses. but was also aware of this possibility, because wherever i got hidusbf from back in that day there was an actual warning about this.

LordOfMice, 2019-01-16:
1. It's overclocking. NO GUARANTEES AT ALL.
2. You don't understand well what you are talking about and what this software really does.

swineadam, 2019-01-17:
sorry... but yeah, i don't fully understand the software. but i do know it can be used to set the polling rate for your mouse. and i do know that doing so could eventually burn out the laser on an optical mouse. this is all i was trying to say, that increasing the poll rate can do this.

as previously mentioned, wherever the original location of this software was back some while ago actually warned about this. not trying to be rude or anything, i just thought it would be friendly enough to warn people about it. oh wells, no worries... is your call, i am not trying to be pushy.

swineadam, 2019-01-17:
anyhow, thank you for your time. take care! :)

closed 2019-05-04 by LordOfMice