LordOfMice/hidusbf #6: Windows 10 1903 can't overclock logitech MX518(downclocking works!) on AMD platform

dongerlel, 2020-03-08:
My specs:
CPU: ryzen 3600
MB: asus b450 pro gaming
Mouse: logitech MX518

Steps i did:
1. launched SETUP.EXE from hidusbf-master\hidusbf\DRIVER
2. pressed install service
3. check filter on device, change to 500 and restart
4. the mouse stays on original 125hz

It is possible to downclock to 62 and 31 with no problem though. Please help. It was working just fine with steps just like above on my intel platform with the very same mouse. So i am 100% sure the mouse can go 500hz.

[image]

LordOfMice, 2020-03-08:
You have to use HIDUSBFN.ZIP package in addition.

As readme.md recommends:
>Recent info https://www.overclock.net/forum/375-mice/1589644-usb-mouse-hard-overclocking-2000-hz.html (the end of the first post and recent posts in thread)

closed 2020-03-08 by LordOfMice

dongerlel, 2020-03-08:
I've been struggling with this the whole day, can you post additional steps i need to take? Can not work it out from readme

LordOfMice, 2020-03-09:
Yes, if you use Test Mode is so simple as copying.
If you want to run on non Test Mode - then you will find link to 4 step by step manuals on link above and much more.

dongerlel, 2020-03-09:
For people from the future:

https://github.com/MineSweeper73/1000Hz-Polling-Hack-1903