LordOfMice/hidusbf #7: Controller oc issues

gambl3rkk, 2020-04-16:
Hi, I used your program to overclock my ps4 controller, I feel it more responsive, and I can see the input delay went from 3.5 to 0.5 ms, but now I would like to go back to my original input delay because I feel that my sensitivities in game changed and I am not used to play on a too responsive controller.
I have tried the reverse OC, and to install the HIDUSBU file in the folder, I also tried to unistall the controller from device manager but I still feel the overclock. DS4 Windows says that I have 3.5 ms input delay now. Is there a way to remove the oc for sure?

LordOfMice, 2020-04-30:
All you have to do in recent version of SETUP - just disable checkbox "Filter On Device" (and Restart if you wish to apply changes immediately). That's enough and working always.

>I also tried to unistall the controller from device manager but I still feel the overclock. 
"Feel", "Suppose", "Believe" is WRONG terms in such situation. Loading of driver is exact fact and checking of that fact is available.

closed 2020-04-30 by LordOfMice

itzcee, 2020-11-22:
I get an error message when trying to over clock my controller.  A y suggestions?

LordOfMice, 2020-11-23:
1. "Error message" is not descriptive and "overclock" too (overclocking is process consisting of few actions and is not one action). I did this and got that - this is right issue reporting.
2. *Necroposting* in *closed* issue - it is not good idea at all. Create new.

SSJGibby, 2021-02-09:
Hey man I downloaded your program and it worked perfect the first time but when I stopped borrowing my buddies PC and got my own I haven’t been able to Overclock my controller since. It says there’s an anti-cheat blocking it or my computer has a anti-virus or whatever Installed that won’t let the program modify any devices. Then the bar turns red and I can’t use the controller anymore. Got any advice or help for me? Or maybe some ideas. I’ve been researching for months at this point.

soopyy, 2021-02-17:
@SSJGibby i also have that issue @LordOfMice could u respond asap i dont know how to fix this filters dont work at all Soop#6618 on discord if u could message me about it or respond here.

hotagu, 2021-09-25:
@SSJGibby @soopyy  
I have running to that issue did you find a way to fix the problem ever since?

gambl3rkk, 2021-09-25:
You have to uninstall the drivers, the program and the folders of the program if they remain in your pc. Then you should be fine

hotagu, 2021-09-25:
@gambl3rkk  Thanks I uninstalled HID - driver from device manager and restarted my computer. It worked!

nesbe09, 2021-10-02:
I have an issue with my controller after overclocking, it has kind of a drift that goes up but it is not constant. Also the controller is a brand new ps4 which dont have any drift I check it on ds4 windows. Any possible cause ?

colehiemstraa, 2021-12-13:
> @gambl3rkk Thanks I uninstalled HID - driver from device manager and restarted my computer. It worked!

I have this issue but am not sure what you mean by uninstalling HID. Please help.

BraydanC, 2022-03-31:
A few times now I've had to "reinstall windows" (just running the media creation tool) in order to be able to utilize the HID toolkit to overclock my controller. Otherwise when attempting to reinstall the service and check the filter box, I receive the error 193 message that something (AV, anticheat, broken service, etc) is absent or wrong and prevents filter service from running, and device will stop functioning after restart with high probability. 

In your experience, is that typically a game's anti-cheat stopping the service from running? Or is there anything I can do to stop that error from occurring every few weeks? I only really play Call of Duty, which they implemented a new anti-cheat, so figured that's probably what was occurring but just wanted to inquire further. Thanks for your time.

LordOfMice, 2022-03-31:
>I receive the error 193 message

1. I assume you have to read PINNED issue with that error
2. Necroposting in closed issue - it is not good idea at all. Create new. VERSION 2.0: https://github.com/LordOfMice/hidusbf/issues/7#issuecomment-731994044