LordOfMice/hidusbf #11: Restart Function

moheat, 2020-08-03:
I haven't been able to use the Restart Device function since a few days. I suspect it's because of the new windows update because I reset my computer without installing anything else (an anti-virus or otherwise) and the error still shows up. 
[image]

LordOfMice, 2020-08-20:
Try to run SX64.EXE directly (as Admin, of cause) and check errors. 
And you can use 100% working workaround described in readme - detach and attach device or reboot.

AimforWC, 2020-09-16:
Hey LordofMice my overclocking is very laggy and i turned secure boot off and it still feels laggy. I really need help.

LordOfMice, 2020-09-17:
>overclocking is very laggy
because it is overclocking, isn't it? this is device dependent whether you get something useful or no.

closed 2020-09-24 by LordOfMice