LordOfMice/hidusbf #16: Getting an error when trying to "filter on device" and needing to disable driver signature [Official GameCube Controller Adapter]

Stenaven, 2020-10-12:
### I am trying to "overclock" the polling of my GameCube Controller adapter (WUP-028) to 1000Hz. 
The default is 125Hz (8 ms). The error I am getting when I activate the "_filter on device_" option for the adapter is “_Error trying to test loading of filter service: 577_”

In order for me to fix this error, I had to disable Driver Signature and disable Secure Boot in my BIOS by deleting "Platform Key" (PK). I also have to keep them disabled or my adapter won't function properly (of course, I can always revert all these changes, but that defeats the goal of using hidusbf). 

Ultimately, **is there a way to fix/bypass having to go through all of this just to edit my adapter**? 

The goal is to be able to play competitive Super Smash Bros. Melee, both on the Slippi Dolphin emulator and play online, without having inputs dropped/delayed from the default 125Hz polling rate of the adapter (stated by Arte's google doc and tweet). 

By deleting the "Platform Key," disabling Secure Boot, and disabling Driver Signature, it gives me "_security risks_", and the corner of my desktop has "Test Mode" dialogue (pic below). I would like to avoid these issues and not have worry of "_security risks_" and whatnot. 

I hope I can get help with this. I hope I explained everything well. Thank you for your time.

If it matters, I am on Windows 10 and have an ASUS motherboard/BIOS.

### Below are sources to what steps I followed and the information I had.

**[-] Overclocking the adapter**
https://docs.google.com/document/d/1cQ3pbKZm_yUtcLK9ZIXyPzVbTJkvnfxKIyvuFMwzWe0/edit#

**[-] Disabling Secure Boot**
https://www.technorms.com/45538/disable-enable-secure-boot-asus-motherboard-uefi-bios-utility

**[-] Disabling Driver Signature (Solution 2, recommended on page 7 of the linked google doc)**
https://www.wintips.org/how-to-fix-windows-cannot-verify-the-digital-signature-for-this-file-error-in-windows-8-7-vista/#solution-2

**[-] Arte's Twitter post (the one who suggested using hidusbf for Melee and made the google doc)**
https://twitter.com/SSBM_Arte/status/1313616578571837440

**[-] Info about the _Official_ GameCube controller adapter**
https://wiki.dolphin-emu.org/index.php?title=How_to_use_the_Official_GameCube_Controller_Adapter_for_Wii_U_in_Dolphin

**[-] The github for the Slippi-Dolphin emulator**
https://github.com/project-slippi/Ishiiruka

**[-] My devices in hidusbf (if it helps/matters)**
[image]

**[-] The message displayed deleting the PK and disabling Secure Boot**
[image]

closed 2020-12-23 by LordOfMice

jrichbestballa, 2021-01-15:
Hey Lord Mice,
I'm trying to overclock my xbox controller and keep getting this notification I was seeing what i can do to fix this problem 
[image]

LordOfMice, 2021-01-15:
Yes, it is exactly problem of this issue. But you have to read answer, i assume. Answer IS here!

jrichbestballa, 2021-01-15:
Ok thanks is there a risk taking off the secure boot ? And do you have to play with it off everytime ?

LordOfMice, 2021-01-15:
Once again, answer IS here. Yes, with Secure Boot off you (to say exactly - code on your computer with enought rights) can load drivers without Microsoft's signature (but not without any signature, of course). It is definitely more risk. And that is QUESTION of this issue starter. Try to read and understand.

>And do you have to play with it off everytime ?
I don't have PC with Secure Boot right now, so it is definitely off forever for me.

eduardo075, 2021-04-11:
the disabled drive subscription does not allow you to open fortnite

LordOfMice, 2021-07-21:
We have money but need someone who can organize signing process.
Current state (and problem):
https://www.overclock.net/threads/digitally-signed-sweetlow-1000hz-mouse-driver.1597441/post-28818893

LordOfMice, 2022-03-11:
Ok, as we have first right question about the option to load HIDUSBF with Secure Boot On and this option independently confirmed now:
https://github.com/LordOfMice/hidusbf/issues/132

suufferPNG, 2022-08-15:
I havent seen this brought up cause I know that you need secure boot off but an issue im having on windows 11 is I need it in order to valorant to run. I want to be able to turn my gamecube controller adapter up to 1000hz for slippi online but those two dont allow for the other to work.

LordOfMice, 2022-08-17:
You have to read answers in this issue. More i can say - the answer to your question right above!

JB0408, 2022-12-07:
I've stupidly clicked yes by accident on the error 577 now my ps4 controller won't switch on. Please HELP!!!!

LordOfMice, 2022-12-07:
>Tried this and it still won’t power up or register when plugged in

JFYI:
https://github.com/LordOfMice/hidusbf/issues/190

LordOfMice, 2022-12-11:
>don’t understand what I should do

>Show me the [CopyIDs] result.

LordOfMice, 2022-12-31:
More recent version of drivers signed by third party certificate:
https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/post-29088655
N.B. - some AVs definitely do NOT like this certificate

thirdeye188, 2023-12-19:
Yo I need help trying to uses hidusbf like it don't let me use my controller it turns red when I put the files on and I put 8000 for the ps5 controller I need help if u can help me I'll send my discord so I can screen share

hanspeter123187, 2024-03-19:
So i deleted hidusbf.sys from System32, then i realised it aint worth it to damage my pc for an overclock. How can i get the original hidusbf.sys file back?

LordOfMice, 2024-07-30:
The latest version of drivers are signed by Microsoft Attestattion Signing. Thanks to Battle Beaver Customs http://www.battlebeavercustoms.com/