LordOfMice/hidusbf #25: setup.exe is deleting itself

oskarjhk, 2020-12-22:
whenever I press install service setup.exe closes and deletes itself. I have disabled secure boot and am running on 64-bit Windows 10.

oskarjhk, 2020-12-22:
https://user-images.githubusercontent.com/68066983/102938753-d42d5c80-44a4-11eb-8f68-b21c12e3ebef.mp4

LordOfMice, 2020-12-23:
it is VERY hard to delete executable module on modern Windows itself. check you AV or something like this.

closed 2020-12-23 by LordOfMice

oskarjhk, 2020-12-23:
> It is VERY hard to delete executable module on modern Windows itself. Check your AV or something like this.

what is an AV

LordOfMice, 2020-12-23:
Google "AV acronym", there is the first definition. Is is not smart question.

oskarjhk, 2020-12-23:
> Google "AV acronym", there is the first definition. It is not smart question.

ok, but how would I check whether the antivirus is affecting the software you didn't respond to that part of the question. I've googled it and it says to check here to allow any programs through the firewall but I don't know what setup.exe would be called here  

[image]

LordOfMice, 2020-12-23:
If you can not distinguish between anti-virus and firewall I have one clear recommendation - or educate yourself or find someone who can help you.

oskarjhk, 2020-12-26:
> If you can not distinguish between anti-virus and firewall I have one clear recommendation - or educate yourself or find someone who can help you.

I DONT HAVE AN ANTIVIRUS THE ONLY THING WHICH COULD BE BLOCKING YOUR SOFTWARE IS MY FIREWALL

supernayru, 2021-08-06:
lord of mice you are no help at all

supernayru, 2021-08-06:
> If you can not distinguish between anti-virus and firewall I have one clear recommendation - or educate yourself or find someone who can help you.

you fucking suck at helping people and expect people to just know things
work on your patience you stupid faggot.