LordOfMice/hidusbf #26: Controller softbricked?

baccpack, 2020-12-24:
Got a brand new PS4 controller that I was gonna use for CoD and after filtering my device it went red and blank on hidusbf. Everytime I plug it in it windows says "USB not recognized." When I try to go back into the program I get error 577 and can't filter. How can I fix my controller? PLEASE HELP!

LordOfMice, 2020-12-26:
Read readme. It's so easy.

closed 2020-12-26 by LordOfMice

LordOfMice, 2020-12-30:
Ok. But your have one of two possible problems - or something broken in your operating system or your controller is HARDbriked. After deleting device this software no more loaded for this device. You can check this in device manager:
[image]