LordOfMice/hidusbf #27: error 577

st0nned, 2021-01-04:
I want to solve error 577, but I think I need to enter the BIOS, only that my bios is blocked, how do I solve it in some simple way?
my controller stops working after I put the filter device

LordOfMice, 2021-01-04:
What did you mean "blocked"? If you have modern BIOS (and 577 error definitely has its source from TOO modern BIOS) you can try to enter BIOS setup through standard modern OS (the second source of problem) possibility (google "boot to uefi windows 10").

>some simple way
There is no simple (means right ) way but there IS some way probably. I can give you link for this way, of course, it is not secret but rather very little-known. But i cant help more because I can not test it myself fully (i don't have PC with Secure Boot right now). It is working, but IDK is it working in your situation - you are the first ;)

Plexi0, 2021-01-09:
Hi! I am currently experiencing a similar issue. I have not yet tried to boot my computer into Secure Boot. The issue only seems to occur when trying to overlcock a Xbox one controller from what I have noticed and seen online. When I try to apply the filter it gives me and other people this error. I will link a screenshot of the error, I am not sure if it is something that could potentially be resolved/fixed or if it's something we need to do on our end to make it work like booting into Secure Boot.

https://gyazo.com/b5234956082167862d446b3d38bc3c38

LordOfMice, 2021-01-10:
>if it's something we need to do on our end to make it work like booting into Secure Boot
I answered this question and it's pinned right now:
https://github.com/LordOfMice/hidusbf/issues/16

LordOfMice, 2021-01-14:
There is one unaswered question for me always - for what purpose such kind of persons asking for help and leave topic after that forever?

closed 2021-01-14 by LordOfMice