LordOfMice/hidusbf #28: Error trying to test loading of filter service: 1060

CrazyCherry7, 2021-01-05:
"The specified service does not exist as an installed service" Something (Windows 10 x64 with Secure Boot, antivirus, protection/anticheat engine, broken installation of service, absent or wrong

C:\WINDOWS\SYSTEM32\DRIVERS\HIDUSBF.SYS, hidusbf.sys from hidusbfN.zip without Tes Mode on x64, enforced kernel code loading policy, etc.) prevents filter service from running; device will stop functioning after [Restart] with high probability, continue Setup?

Any suggestions on how to fix it. I was OC my Xbox elite series 2 controller, "filter on device", then this appeared. I've been using this software for almost a full year now and never have I ever ran into this issue/error.

LordOfMice, 2021-01-05:
try to install service once again, of course. as readme recommends through setup->[Install Service] or hdusbf.inf->Install.

closed 2021-01-05 by LordOfMice

tryhardpe, 2021-09-23:
CrazyCherry7  
You could solve it I have that same problem could you help me please this is my discord Tryhardd # 9058 add me I will reward you

LordOfMice, 2021-09-24:
No, i can not help you in this form.

PoeticJustis, 2025-05-13:
I’m having this issue. I have been trying to fix it for hours. Please help!

PoeticJustis, 2025-05-13:
I’ve uninstalled, reinstalled, disabled secure boot, enabled test mode, disabled memory integrity, tried manually installing it through device manager, and nothing is working. Setup.exe is running as administrator and it is the correct file location. I cannot figure this out.

LordOfMice, 2025-05-13:
>tried manually installing it through device manager

What?