LordOfMice/hidusbf #30: Secure Boot

kyledalt, 2021-01-14:
Hi mate, amazing program, I was wondering if I could get some assistance from you if you don't mind.
I previously couldnt get my xbox remote overclocked, but after disabling secureboot it worked!

Can i turn secure boot back on?
Will i notice any performance issues or any other issues with secure boot off?

Thanks

LordOfMice, 2021-01-14:
>Can i turn secure boot back on?

No.

>Will i notice any performance issues

No.

>or any other issues with secure boot off?

https://github.com/LordOfMice/hidusbf/issues/16

closed 2021-01-14 by LordOfMice

SergioVB-99, 2022-03-18:
WINDOWS 11 forces me to use secure boot to play valorant, for example, so the program does not allow me to overclock my controller. What I can do?