LordOfMice/hidusbf #38: Alerts when overclocking is reversed

johnmgarretson, 2021-02-08:
Overclocking can be reversed by system updates, etc. It would be great to have a desktop app that can let me know if overclocking has been reverted so I can go back in and turn it on.

I would pay $100 to sponsor development of such a feature. If interested, just reply with a Bitcoin address and I will send the sponsorship money before it has been developed or released.

Thanks for your software!!!

LordOfMice, 2021-02-08:
My friend, all you need to check that driver is working at least for one device is test that driver is loaded and its service is running.
Put 
https://github.com/LordOfMice/hidusbf/blob/master/check_hidusbf_running.cmd
in autorun and it will stop with warning in wrong case

>I would pay $100 to sponsor development of such a feature.
better to use this money for new sign of drivers:
https://github.com/LordOfMice/hidusbf/issues/16#issuecomment-707788749
;)

closed 2021-02-08 by LordOfMice

johnmgarretson, 2021-02-09:
Nice, thank you!

LordOfMice, 2021-02-09:
P.S. And, of course, if you use patcher trough atsiv - don't forget to check that patcher PATCHED something useful, now it's possible.