LordOfMice/hidusbf #40: My controller won’t Overclock

SSJGibby, 2021-02-09:
I have either an anti-cheat or anti-virus that won’t let me Overclock my controller. When I do it just disable the device. Is there a setting or something I need to switch? I’ve tried all that I know to try. Googled it a 100 times. Can’t seem TTK get it to work. Bar just turns red and I can’t use the controller anymore till I revert the changes made. Any help or advice I’d appreciated.

LordOfMice, 2021-02-09:
I think Setup has to show some error code and description when you try to install filter on device. But you are omitted this information for unknown reason.

SSJGibby, 2021-02-09:
I’m currently in bed and not at my pc so I didn’t know the error code. But
apparently it’s a problem with the secure boot cause someone has already
had the same issue before. It’s a thing with Asus motherboards apparently.

LordOfMice, 2021-02-09:
>it’s a problem with the secure boot cause someone has already had the same issue before
why are you asking for help if you know the answer?

closed 2021-02-09 by LordOfMice

SSJGibby, 2021-02-09:
I’m new to all of this and don’t know how to work whatever this forum is. I
posted them I saw later on that it was already talked about.

handyprod, 2021-11-02:
[image]
 hi , ive installed services i dosent show anything like if something really installed and dosnt show any error mesage neither , but my bi interval is still 5

LordOfMice, 2021-11-03:
Restart...

Seniic, 2022-07-18:
Can someone please help me?
I have windows 11 and GitHub to overclock my ps4 controller isn’t working 
It appears red bar and I can’t use the controller.
I have input delay around 1-1,5 and I want it to put to 0-0,5% 
What should I do?

LordOfMice, 2022-07-19:
You should read readme (more i can say - WARNINGS in readme) and pinned issues.