LordOfMice/hidusbf #41: Worked.. for a day

SweetDollaT, 2021-02-09:
I followed a tutorial on this over clock yesterday and all of yesterday it worked and felt amazing. I booted up to warzone today and my game felt super laggy so I checked the program and it said my controller was 1000 to 1 and every setting was the same from yesterday. Idk how its not working today every setting is right? any help would be GREATLY APPRECIATED

SweetDollaT, 2021-02-09:
Running a Battle Beaver dual shock 4 btw

SweetDollaT, 2021-02-09:
I also don’t believe any files are missing and I haven’t gotten an error code either.

LordOfMice, 2021-02-09:
If you have full speed device and got bInterval=1 in setup after restart with Rate=1000 then software is working as expected. It's overclocking, something can go wrong.

closed 2021-02-09 by LordOfMice