LordOfMice/hidusbf #42: hidusbf.inf won't install

Gingynator, 2021-02-09:
It's very odd. I've been using this program for about 6 months when I replaced my controller. My new controller behaved erratically with inputs that would come in half a second late (typical for a ds4 on bluetooth connection) so I uninstalled hidusbf for a fresh reinstall. The issue is now when I run setup.exe and I click "Install Service" I get an error message that simply says "Installation Failed". I get the same issue when I try to manually install HIDUSBF through right click -> Install. Any help would be appreciated.

LordOfMice, 2021-02-09:
if you will execute this command with admin rights:
sc create hidusbf type= kernel start= demand binpath= system32\Drivers\hidusbf.sys
(spaces ARE matter) and copy appropriate hidusbf.sys (from amd64 folder, probably) to %windir%\system32\Drivers\
then you get exactly what installation of hidusbf.inf does.

closed 2021-02-09 by LordOfMice

Gingynator, 2021-02-09:
I verified that my account has admin rights (which it would since it's the only user) and it still wouldn't work. I copied and pasted your command in the command prompt. 
[image]

Gingynator, 2021-02-09:
The hidusbf.sys file is already in there, along with another one from 2019. Should I delete those and replace them?
[image]

Gingynator, 2021-02-09:
I learned that I can run the command prompt as an admin. Phew. I tried again and this is what I got:
[image]

LordOfMice, 2021-02-09:
>The hidusbf.sys file is already in there, along with another one from 2019. Should I delete those and replace them?
There is only ONE hidusbF.sys. "another one from 2019" is system file and its name is NOT hidusbF.sys.

>I tried again and this is what I got
ok. reboot. you install hidusbfU.inf - this is file to DELETE service. After that try to install hidusbf service by usual way through setup->[Install Service]. Be careful - one letter difference is MATTER.

Gingynator, 2021-02-09:
I deleted the hidusbf.inf and it's working now as indicated by the filter -> yes, rate - > 1000 (what I want), and bit interval - > 1. I don't know if this is something interesting that will help you in the future, but I still got the error message with "install service", but it works now so I'm not complaining.

Gingynator, 2021-02-09:
Also I really appreciate all of your help :)

alighasoub-web, 2025-09-13:
> if you will execute this command with admin rights: sc create hidusbf type= kernel start= demand binpath= system32\Drivers\hidusbf.sys (spaces ARE matter) and copy appropriate hidusbf.sys (from [...]

I just created this account to just thank you so much because of this command , my overclocking worked i tried all options and this one that worked for me . I had the 1060 error