LordOfMice/hidusbf #47: Just a question

User986514, 2021-03-04:
How come certain people don't have to disable their secure boot and it just works off bat? Is it to do with pc parts maybe? Idk. Also, is it safe to disable secure boot and is that the only way I can fix the red bar?

User986514, 2021-03-04:
Also, let's say I do end up turning off secure boot.. will it work just as fine as it would for people who have done this without an issue?

LordOfMice, 2021-03-04:
https://github.com/LordOfMice/hidusbf/issues/16

>is that the only way I can fix the red bar?

No. The second if you will find me US resident which is ready for process to get EV sign and Microsoft Verified Account for new signing. We already have sponsor (and money) but not the organizer.

closed 2021-03-04 by LordOfMice