LordOfMice/hidusbf #48: Readme clarification?

Kein, 2021-03-05:
>This driver/program is intended to "overclock" USB mice **(and other devices)** 

>Program changes **only mouse rate**, not touching other devices on USB.

???

LordOfMice, 2021-03-06:
Check is it correct now.

closed 2021-03-06 by LordOfMice

Kein, 2021-03-06:
What happens if a device already has vendor drivers installed?

LordOfMice, 2021-03-06:
There is no difference MS or vendor drivers used.

Calzzy, 2021-03-22:
Sir how do i fix this
when i type Atsiv.exe -f hidusbf.sys and enter this shows up
0xc0000001, failed to load Atsiv service

LordOfMice, 2021-03-23:
It's offtopic for this issue.
But you have to 
1. run atsiv with elevated privileges, of course 
2. carefully read info from link to OCN. In addition to antiviruses some game protection engines prevent atsiv driver loading. And there is workaround for some (not all) cases.