LordOfMice/hidusbf #49: Error when trying to overclock my ps4 controller

app4k, 2021-03-09:
When I click "install service" I get the two warnings attached below. Any ideas on how to fix this?
[image]
[image]

LordOfMice, 2021-03-09:
When you [Install Service] you can not get these messages. You will get this messages on [Filter On Device] exactly if you didn't Install Service first.

>Any ideas on how to fix this?
read readme, of course

closed 2021-03-09 by LordOfMice

LordOfMice, 2021-03-10:
readme:
In case of HIDUSBF service wasn't installed through SETUP.EXE (button "Install Service") you may try to do this by OS standard possibility: mouse right button click on HIDUSBF.INF -> Install
And, of course, unzip all package before anything else.