LordOfMice/hidusbf #51: Hey

darksinthe, 2021-03-14:
a week or so ago I accidentally clicked the wrong device when trying to do my controller (ive done my dualshock controller with this program a few times beforebefore) and changed the bitrate, nothing happened. Then i realized i had chosen the wrong device. I didnt notice anything wrong, so I ignored it and continued to OC the controller. Today i realized my Valve Index VR headset isnt doing anything when i plug it in, Soooo i must have accidentally picked my VR headset instead of my controller and tried to change the bitrate. Now I cant get it to work at all. is it broken?

Chaython, 2021-03-14:
Please don't just do this.
Actually say the issue, and how you fixed it. So people can search it in the future.

LordOfMice, 2021-03-15:
>Actually say the issue, and how you fixed it. So people can search it in the future.

99% of "issues" here is just RTFM problem. I can say this project is reference example.

closed 2021-03-15 by LordOfMice