LordOfMice/hidusbf #53: Failed to install service

TheRealGeDev, 2021-03-21:
[image]
can't install the service for some reason. 
win10 1909 / no antivirus / secure boot off / no virus of some sort / no anticheat

btw using this version https://www.youtube.com/watch?v=DtY_3sAJJnE the service installs but the device manager says that it's corrupted / the reg key is fucked up

LordOfMice, 2021-03-21:
577  means secure boot on with version from hidusbf.zip or you try to use version for test mode (from hidusbfn.zip) under non test mode

closed 2021-03-21 by LordOfMice

TheRealGeDev, 2021-03-22:
secure boot is off and i m not using the test mode one

LordOfMice, 2021-03-22:
So answer is simple - you can't get this error if you have ordinary system. Of course, I know how to get this error on system without Secure Boot and (with high probabylity) how to load my driver on system with Secure Boot - but this is not _ordinary_ knowledge.