LordOfMice/hidusbf #54: Help please, OC ds4 controller, error 193

Beaan, 2021-03-23:
Hello, i am trying to overclock my ds4 controller but when i click "filter on device" i get an error 193 message
[image]
i have disabled secure boot, and disabled my anti virus, and tried all different USB ports (including USB 3.0) still with the same issue. I have also tried to reinstall my controller drivers and i do not have ds4windows installed or anything similiar. When i click on "install service" or install it via the HIDUSBF.INF, i just see this small transparent pop up that lasts a milisecond so i dont know if its installed correctly? 
[image]
Does anyone know a solution? i cant think of any at all.
Thanks

LordOfMice, 2021-03-24:
Try to uninstall service through hidusbfU.inf apply (Right Click->Install) or through console command (with elevated rights):
sc.exe delete hidusbf
then reboot, then install hidsubf as usual

Beaan, 2021-03-24:
just tried it, and i still get the same error message, i dont know what else to do

LordOfMice, 2021-03-25:
ok, 
1. show me  output for
sc start hidusbf
in elevated rights console
2. check event viewer for System too - what is reaction (event) for above command?
3. what is OS you are using? exact version, of course
4. send me dump (.reg) for 
HKEY_LOCAL_MACHINE\System\CurrentControlSet\Services\hidusbf

Beaan, 2021-03-25:
ok so im not really tech savvy but ill try my best
1 - i think you are referring to cmd command in admin mode right? if so, this is what i see: 
[image]
2 - i saw three events:
[image]
3 - Windows 10 pro version 10.0.19042 build 19042
4 - im not sure how to do this, i think i done it? but i had to convert it into zip to attach here. 
[file.zip](https://github.com/LordOfMice/hidusbf/files/6207739/file.zip)

many thanks for your help on this dude appreciate it

LordOfMice, 2021-03-26:
Ok, you made this mistake:
https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/page-34#post-28214614
And yes, it is referenced in manual, you can find it youself...

closed 2021-03-26 by LordOfMice

Beaan, 2021-03-26:
I dont mean to sound like an idiot but ive spent the last few hours (and days before opening this issue) trying to understand all of this, but im really sorry im not that smart when it comes to computer stuff, as far as i can tell i have a different architecture and so i need to correct for this? 
I want to use the "atsiv" method and not the test mode one (because i play online games and i read test mode may cause issues with anti-cheating etc) and atsiv is a way to get unsigned drivers installed correctly but i dont know how to use it or where to get it from?
Please forgive my ignorance, from the link you sent, that is for win 10 2004? i have win 10 20H2
Ive read the "readme" files in the hidusbf and hisusbfn, cant seem to understand what to do though. 
i know this is really simple and obvious for you but id appreciate if you could run me through what i need to do

LordOfMice, 2021-03-27:
1. Uninstall first. For completeness - check that hidusbf.sys deleted from %windir%\system32\drivers
2. Get hidusbf.zip and unzip it - with folders. Do NOTHING after that.
3. Run setup from driver folder -> install service and try to decrease rate to 31 or 62 on USB mouse (i suppose you have USB mouse?) Check that rate is decreased. This is exactly what readme recommends to do first.
Only after that try overclocking.

>i have win 10 20H2
Error 193 does not relate to version. We try to solve THIS problem here.

Beaan, 2021-03-27:
Yeah i have a USB mouse, i followed your steps and used the mouse tester from dobragab`s github, and some other online ones to test my mouse polling rate.

after decreasing the rate to 31 and 62 ( i did not tick the "filter on device" box) for my USB mouse, i didnt notice any difference, my mouse was still at around 400-600hz (mainly 500hz) and 2-3ms (mainly 2ms). I tried to overclock it to 1000hz and again no difference. 

although im not getting an error message.

 (if i try to overclock my ds4 without ticking "filter on device" i get this message : 
[image]
)

LordOfMice, 2021-03-28:
>i did not tick the "filter on device" box

But WHY???

Beaan, 2021-03-28:
> 
> 
> > i did not tick the "filter on device" box
> 
> But WHY???

loool because i get error 193 and if i press restart or plug in and out my device, it stops working entirely until i switch USB ports

LordOfMice, 2021-03-29:
I don't understand - if you know what is the source of your problem then why are you asking for help? If you don't know - try to do what I'm asking to do. Am I clear enough?

>i get error 193
Ok, what is the version, date and size of %windir%\system32\drivers\hidusbf.sys after Install Service?

>until i switch USB ports
I recommend you to read readme about how to revert failed installation. It is literally few clicks (or key presses) in Setup.

Beaan, 2021-03-29:
So, when i click on filter on device i get that error but i dont know how to correct it, i couldnt understand the link you sent im not too tech savvy.
[image]
i tried to do everything you said but im still getting error 193.

LordOfMice, 2021-03-30:
Ok. You have 64-bit version of Windows. The last resort - check signature of this file (SHA256 version). Something like this:
[image]

Beaan, 2021-03-30:
ok so i went on digital signatures -> SHA256 version details -> view certificate -> certification path
this is what i see:
[image]

LordOfMice, 2021-03-31:
Something is strange in your system. It's definitely prevents driver loading, but now i don't have any ideas.
I tried to reproduce some situations but usual errors under Windows 10 1809 are 577 and 1275.
P.S. You can try to check with "Disable driver signature enforcement" startup option on booting. This is 100% detection of problem with OS related driver signature checking.

LordOfMice, 2021-03-31:
Ok, I reproduced error 193. It appears when loader doesn't consider driver file as having [portable] executable (PE) format.
With high probability this is some kind of antivirus or something like this.

Beaan, 2021-04-01:
I have malwarebytes installed on my pc, i usually close and disable it before i try to run setup.exe.
What other steps could i do to disable any antivirus thing in my system?

LordOfMice, 2021-04-02:
1. I can not help with other software.
2. "You can try to check with "Disable driver signature enforcement" startup option on booting. This is 100% detection of problem with OS related driver signature checking." I didn't see any reaction.

Beaan, 2021-04-02:
sorry, forgot to reply
2. I tried it, still producing error 193.
I also turned off all my anti virus, didnt work.
I thnk there is something else, maybe the file is damaged or some of it corrupted?

LordOfMice, 2021-04-02:
File from archive can not be damaged. And you already did this - taking files archive (really did, i suppose?) few steps above.

Beaan, 2021-04-05:
Yeah, ok so i just factory resetted my pc and it seemed to have fixed the problem i think. 
I am able to overclock or underclock my usb mice now. 
Thanks for all the help man and great program

LordOfMice, 2021-04-05:
It is solution, but it would be more interesting to know what is the real root of the problem, of course.

DanPrDz88, 2021-07-20:
Have you solved this issue yet? I'm having the same problem but with a PS5 controller.

LordOfMice, 2021-07-21:
It is not issue of this software so i can not solve anything. And root of problem is unknown now so i can not give any workarounds except one that helped.

ConallWilliamson, 2021-12-07:
Finally found a  solution to this: followed the steps on this link and am now able to overclock: https://support.microsoft.com/en-us/windows/a-driver-can-t-load-on-this-device-8eea34e5-ff4b-16ec-870d-61a4a43b3dd5

LordOfMice, 2021-12-08:
>followed the steps on this link and am now able to overclock

Nice finding and thanks for this.
But this option is NOT enforced in default installation AFAIK. And yes, I suppose people have to know what they did in they systems and understand consequences.

motty4, 2021-12-29:
i did all the step and it still cant pass 125 hz i tred lower it  to 32 and 61 and it worked but it still cant pass 125 hz

Coddler-afk, 2022-01-05:
I also had the same problem erro 193 and disabling Memory Integrity worked for me. I would advise turning it back on after the overclock process. Thank you @ConallWilliamson and @LordOfMice 

https://github.com/LordOfMice/hidusbf/issues/54#issuecomment-988298415

Naworm, 2022-08-24:
Turn "Core isolation" off, reboot and the issue should be fixed.

[image]

LordOfMice, 2022-09-01:
>USE NOPATCH VERSION of driver exactly as readme recommends. You can use above mentioned option (Memory integrity) enabled.

I retested this and can say - no, using nopatch version is not enough, you have to disable memory integrity to load any public version of hidusbf.sys

soopyy, 2022-12-09:
how would i turn back on Memory Integrity and delete the hidusbf.sys SweetLow USB Mouse Rate Adjuster i tried looking in device manager and i can't seem to find it do u know the location of where it is cause i would be interested if there is a way to get it back on.

indecipher, 2023-01-26:
Hello, I have Windows 11 and had been experiencing the error 193. 
I was able to resolve the error by adding "UpgradedSystem" or "WhqlSettings" to the registry and disabling Memory Integrity Core Isolation in the Windows Defender settings. However, I did not like the fact that this left my system vulnerable to malware attacks. 
I did some research of what the Memory Integrity Core Isolation setting is used for, and it is used specifically to protect against virtualization attacks. I then researched how to disable virtualization technology on my system. This needs to be done through the BIOS settings and I would recommend using a search engine to figure out where this setting is for your system. 
So after that, I disabled "Intel Virtualization Technology" in my BIOS settings. After a reboot, the Memory Integrity Core Isolation setting was removed from my Defender settings and I was able to install the hidusbf driver without disabling any security settings.
Hope this helps anyone who was experiencing this issue. Note that you still need to add either of the registry values listed.

LordOfMice, 2023-01-27:
>protect against virtualization attacks
>I disabled "Intel Virtualization Technology"

You win in one area and loose in other:
Virtualization-based Security (VBS)
https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-vbs

indecipher, 2023-01-27:
VBS has always been off for me. I have read that this is best to leave off for better gaming performance. 
Enabling Memory Integrity (HVCI) does not allow me to use the hidusbf driver. So turning off Intel Virtualization Technology would be best, since it is not being utilized for any functionality.

LordOfMice, 2023-01-27:
> I did not like the fact that this left my system vulnerable to malware attacks.
> I have read that this is best to leave off for better gaming performance.
Hmm, I smell something. Eh, "security - performance trade-off"

mrlance2023, 2023-05-23:
> Finally found a solution to this: followed the steps on this link and am now able to overclock: https://support.microsoft.com/en-us/windows/a-driver-can-t-load-on-this-device-8eea34e5-ff4b-16ec-870d-61a4a43b3dd5

I literally made an account right now to say thank you for solving my problem! The core isolation memory integrity and vulnerable driver block list was preventing me from installing the service. I literally wiped my computer four times with a fresh install of Windows 11 to see if there was something not installing properly lol
[image]

Impact009, 2023-08-30:
> > The core isolation memory integrity and vulnerable driver block list was preventing me from installing the service.

Win 11 Pro 22H2 22621.2134 here. I only needed to disable Memory integrity. The service installed fine with Microsoft Vulnerable Driver Blocklist enabled.

74lobster, 2023-09-03:
> > > The core isolation memory integrity and vulnerable driver block list was preventing me from installing the service.
> 
> Win 11 Pro 22H2 22621.2134 here. I only needed to disable Memory integrity. The service installed fine with Microsoft Vulnerable Driver Blocklist enabled.

Hi, but once installed the hidusbfu.inf driver, is it possible to you re-enable the Windows 11 core isolation integrity or it gives driver error ?

It's worth the pain install the hidusbfu.inf driver for the Xbox PC gamepad or is it useless ?

74lobster, 2023-09-03:
> > I did not like the fact that this left my system vulnerable to malware attacks.
> > I have read that this is best to leave off for better gaming performance.
> 
> Hmm, I smell something. Eh, "security - performance trade-off"

Hi, it's worth the pain install the hidusbfu.inf driver for the Xbox PC gamepad or is it useless because this gamepad is not affected by this setting ?

LordOfMice, 2023-09-04:
>is it possible to you re-enable the Windows 11 core isolation integrity

Yes, of course.

>for the Xbox PC gamepad or is it useless because this gamepad is not affected by this setting ?

1. The answer is simple - try and tell us. Some XBox devices are definitely overclockable but you haven't told much information about your device.
2. It is not good place to ask such question in this issue as it is off-topic here.

74lobster, 2023-09-04:
> > is it possible to you re-enable the Windows 11 core isolation integrity
> 
> Yes, of course.

Could you please link me the certified version of hidusbfu.inf driver ?

Every time I tried to install the uncertified driver, the pc showed an error message and on the next restart the wireless mouse no longer worked and to get it working again i had to connect it to a different usb port.
It's normal ?

Thanks for your patience.

wsm78, 2023-10-14:
> Could you please link me the certified version of hidusbfu.inf driver ?
>
> Every time I tried to install the uncertified driver, the pc showed an error message and on the next restart the wireless [...]

Bumping this - I cannot re enable memory integrity with Win 11 stating hidusbf.sys SweetLow being an incompatible driver. Is there any way around this? Downloaded as of today (10/14/23) x64.

ottomhorh, 2024-01-23:
https://youtu.be/xKGsk_6FTUg?si=3Kks2WLZR0ay2rmH i been looking to fix this problem for 30 mins after coming across this video prior before the 193 error i was stuck on 1060 error for 2 months until i gave it another chance and tried fixing it and succeeded then successfully found a fix right after 1060 for 193. Hope this helps out. :)

Ironyyy, 2024-02-13:
Well I am having the same issue except my computer doesn't even have core isolation. Some reason windows doesn't even allow me to have device security.
[image]
I've done everything other than disabling core isolation because I literally cannot disable it.

Ironyyy, 2024-02-13:
I just tried to fix the standard hardware security issue and I found out that my bios doesn't even have an advanced settings tab. Its just permanently stuck in EZ mode.

nanoo117, 2024-03-05:
hello I have the same problem but with my victrix pro bfg controller there is no other software? because I'm at 8ms it's impossible to play. I have windows 11 I do not have or cannot access core isolation please help me (I cry in the evening)

nav1415-bit, 2025-01-26:
Hi, so just to clarify, you can't activate the filter if I have HCVI enabled? Because Valorant asks me to have it enabled. If there is a way to have the filter please let me know.

LordOfMice, 2025-05-23:
Readme:

Added 2025/04/09:
The latest NoPatch version of x64 drivers (AMD64_AS) rebuilt to conform Memory Integrity requirements of Windows 10 x64 or Windows 11. Thanks to Battle Beaver Customs https://www.battlebeavercustoms.com/ once again for recurrent signing.

matru, 2025-08-30:
@LordOfMice Is there any reason why the NoPatch version is not the default one? Are we missing anything in the NoPatch version compared to what the default ships with? Thank you for all of your work.

matru, 2025-08-30:
> Yes, of course. The higher USB class rates like 1000Hz for Low Speed USB devices.

Low Speed as in USB lower than 2.0? So all modern controllers, including a Dualshock 4 should work just the same with the NoPatch and default?

royal998king-hash, 2025-09-10:
> Readme:
> 
> Added 2025/04/09: The latest NoPatch version of x64 drivers (AMD64_AS) rebuilt to conform Memory Integrity requirements of Windows 10 x64 or Windows 11. Thanks to Battle Beaver Customs https://www.battlebeavercustoms.com/ once again for recurrent signing.
> 
> P.S. And short explanation why NoPatch version only. It is very simple - because Memory Integrity does exactly what its name means: prevents critical system memory (including code of drivers) from modification (i.e. patching). P.P.S. And default version of driver is NOT NoPatch. To use NoPatch version you have to expressly switch to it . P.P.P.S. And the simplest way to switch is run nopatch.cmd in DRIVER\ folder before installing service.

hi i tried running your software with memory integrity on but it still doesnt work even tho im running nopatch in admin and then opening setup.exe it keeps saying run the nopatch version. may you please give us a full tutorial on how to use the nopatch version step by step, thank you for the amazing software.