LordOfMice/hidusbf #56: Cannot install service

samh1ll, 2021-04-01:


LordOfMice, 2021-04-01:
>doing nothing

Why do you think so? Did you try the next step?

closed 2021-04-01 by LordOfMice

samh1ll, 2021-04-01:
Next step? i have deactivated secure boot etc.

LordOfMice, 2021-04-01:
Yes, the next step, exactly as in readme:

>Then choose the line with your mouse and put a checkmark Filter On Device.

samh1ll, 2021-04-01:
I have done my best to do all steps in the readme. I have no way to check with my controller on Mouse Rate Checker but i did the same steps on my mouse to test and the rate hasnt changed

LordOfMice, 2021-04-01:
>the rate hasnt changed
>all steps in the readme
Not all, of course.

samh1ll, 2021-04-01:
Well evidently yes but i am clearly unable to see where

LordOfMice, 2021-04-01:
Because you are thinking that there is unimportant parts. There is NOT. Try to decrease rate as readme recommends.

samh1ll, 2021-04-01:
I have tried doing the install with the .INF file also and a pop up appears and instantly dissapears also

samh1ll, 2021-04-01:
OK i can go down on my mouse

samh1ll, 2021-04-01:
but i am unsure how to know if i can on my controller