LordOfMice/hidusbf #61: Error 193 while trying to overclock

MrKyle1, 2021-04-19:
I keep getting error 193 while trying to use hidusb to overclock my GameCube adapter. I’ve disabled secure boot, tried uninstalling the adapter drivers and reinstalling them. The only thing recommended I havnt tried is the factory restart that worked for Beaan

ConallWilliamson, 2021-04-19:
I am getting the same error, and I *did* try the factory reset that worked for Beaan and I'm still getting error 193. It would be great if there were a solution to this

MrKyle1, 2021-04-19:
From what I got from the bean thread is that there is some driver that is restricting the reset from happening and acting like an anti cheat??? I think I'm not sure on any of this

LordOfMice, 2021-04-20:
Once again, it can be anti-virus, anti-cheat or any other BS preventing normal driver loading. It is definitely not the problem of my software so you have to find the root of problem yourself.  It would be nice if you could find this root, of course.
JFYI, now I have PC with UEFI Secure Boot and third party anti-virus but successfully check driver even in such case.

closed 2021-04-20 by LordOfMice

MrKyle1, 2021-04-20:
Installing a fresh start of windows is what fixed it for me. Windows can be a bitch

LordOfMice, 2021-04-20:
I know that on default installation all is working fine. It is not new :) What I what to know - who is responsible for screwing things.

LordOfMice, 2021-08-25:
https://github.com/LordOfMice/hidusbf/issues/54