LordOfMice/hidusbf #63: Says it is working but is not

LoganGRL, 2021-04-25:
Hello,

I've been dealing with input lag for a while and a few weeks ago started using this to overclock. It randomly works for about 30 minutes then stops, even while saying it's filtered on device, 1000, and binterval is 1 (I use a PS5 controller).

I downloaded this DSEO from https://www.ngohq.com/driver-signature-enforcement-overrider.html
I put it in test mode, signed the .INF , did a few other steps and the driver worked again, for about 30 minutes. Now it isn't working again.

Is there anything I'm not doing correctly or should do so that it works every time?

spiderman11786RL, 2021-04-26:
i have the same issue

spiderman11786RL, 2021-04-26:
I even tried 2 overclocking programs, (DS4 Windows and AntiMicro) and changed the polling rate to 1000hz on both of them, in hopes to keep the input consistency low. But as always, it works for the first day or couple, then goes back to worse

LoganGRL, 2021-04-26:
I feel your pain, It's really frustrating. Hopefully we can get some tips here.

spiderman11786RL, 2021-04-26:
> I feel your pain, It's really frustrating. Hopefully we can get some tips here.

Yes im pretty sure LordOfMice will reply soon

spiderman11786RL, 2021-04-26:
@LoganGRL btw how do u download that driver overider and what file do i put in?

spiderman11786RL, 2021-04-26:
@LoganGRL i deleted the overclock, deleted winrar, then reinstalled winrar and reinstalled the overclock. It has made it better. I think it was because Winrar wasnt actually working bc it was expired. I just had to delete it then download it again. Search up WinRar download

LordOfMice, 2021-04-26:
1. I assume DSEO and Test Mode is not actual now (i can say "now" == "5 years"). It is not mentioned anywhere in current manuals.
2. If you load driver successfully then it can not "stop". I don't understand what does it mean. Device can stop respond, for example - that i can understand as problem. Filter can be dropped on some OS updates like this https://github.com/LordOfMice/hidusbf/issues/38 issue, but not on the fly. And you description of problem "considerable amount of delay is introduced after a certain amount of time that is worse than it was even before overclocking that makes my game very difficult to play, but there are no errors and everything appears to be fine" looks like some hardware problem. And i can not help is such cases, of course.

spiderman11786RL, 2021-04-26:
@LoganGRL 
do you have a ryzen system?

LoganGRL, 2021-04-26:
Yea that may be the case. Well thanks for taking the time to respond @LordOfMice.

@spiderman11786RL I do have a ryzen system. I just bought my first pc so I could play Rocket League because my old laptop I used for it did not have a graphics card.

spiderman11786RL, 2021-04-26:
I think it might have something to do with ryzen @LoganGRL. Whats ur rl btw?

LoganGRL, 2021-04-26:
Ah well that'd be unfortunate.

I regret making it this lmao but it's
AdmiralCoochie 

What's yours?
I'm in the middle of exams rn but I'm down to run some next week or something.

closed 2021-05-23 by LordOfMice

Scrill69, 2024-02-02:
@LoganGRL did you ever find a fix for this I have the same exact issue