LordOfMice/hidusbf #75: Astro C40 Controller error

jc991, 2021-06-21:
Hello, 

I've been trying to overclock my Astro c40 controller and ran into the 577 error code, so I used the secure boot disable and driver signature disable method to overcome it. I've also saved Setup to run as Admin.

Now after I've clicked filter and 1000, once I click the "restart" button in the "Setup" window I get an error something like the following: 

"If your device does not restart in 10 seconds, you may be running Setup without Admin rights or AV is preventing your Helper module from working properly" 

Any ideas how to resolve this? Would love to get better response time than the default 4ms

LordOfMice, 2021-06-21:
Exactly as described in readme - disconnect then connect device or reboot OS.

closed 2021-06-21 by LordOfMice

LordOfMice, 2021-06-21:
P.S. And you are using the old version of setup because the most recent version has this message:
Device doesn't restart after 10 seconds, you probably have some problem in your system like running Setup without Admin rights or you run Setup with current folder without Helper module (SX64.EXE) on x64 system (possibly from archive directly without unzipping?) or AV is preventing executing of Helper module or something holds this device opened (uses it). You can try to disconnect and then reconnect device or reboot system.