LordOfMice/hidusbf #76: Possible bitcoin virus and mining?

Blankylord, 2021-06-27:
good, sorry if you have already been asked about this, I have tried to inform me I can not find anything.

My problem is when I download Malwarebytes it detects your downloaded file as a virus: 
-MachineLearning/Anomalous.100% (Ad-Aware).

Looking on google it is said to be a bitcoin trojan and mining, my question is the following, is it a false positive or is it true about the virus?

PD: Sorry my bad english.

LordOfMice, 2021-06-27:
>is it a false positive or is it true about the virus?
virustotal.com

closed 2021-06-27 by LordOfMice

Blankylord, 2021-06-27:
I was asking because according to virustotal the malwarebytes database detects it as a trojan, I was just asking if it's true.

LordOfMice, 2021-06-28:
If only one from few tens AVs whines then I assume the answer is obvious.

Chaython, 2021-06-28:
Has nothing to do with mining. It says machine learning! Meaning they fed an "ai" a dataset, and this is dissimilar to any legitimate software they have fed it. [because this is a niche project not a common software.]
Next time search up and report false positives to the av developer https://forums.malwarebytes.com/forum/42-file-detections/

LordOfMice, 2021-06-28:
>Next time search up and report false positives to the av developer

I think false positives is problem of users of such AV, not my. So it is obvious who needs to report.

Blankylord, 2021-06-28:
Thank you very much for not getting angry and for answering me with quite reasonable arguments, forgive me for having doubted, I want to try this program and nowadays you can hardly trust it on the internet, so I hope I have not offended you, I will try the program because you gave me confidence.

PD: Very thanks LordOfMice and Chaython, forgive my ignorance.