LordOfMice/hidusbf #77: error when restarting

spiderman11786RL, 2021-07-03:
When i click restart it says
error!!!
'device doesnt restart after 10 seconds you probably have some problem in your system like running Setup without admin rights or you run Setup with current folder without Helper module (SX64.EXE) on x64 system or AV is preventing executing of Helper module or something or somethign holds this device opened. You can try to disconnect and then reconnect device or reboot system'.
The reason im putting this on here, is because i think it is causing the overclock not to work how do i fix this?

LordOfMice, 2021-07-04:
>how do i fix this?

literally as software suggested you

closed 2021-07-04 by LordOfMice

spiderman11786RL, 2021-07-05:
> > how do i fix this?
> 
> literally as software suggested you

but i do not know how to and what to do

spiderman11786RL, 2021-07-05:
what is av

LordOfMice, 2021-07-05:
>but i do not know how to and what to do

>>You can try to disconnect and then reconnect device or reboot system.

I assume you understand English and some simple terms.