LordOfMice/hidusbf #82: Controller OC becomes Sluggish/less responsive overtime

Christianguerra001, 2021-07-25:
Hello, 

I have an issue with the overclock. Ill do my best to explain my issue but I don’t know if ill be of any help. So my controller after sometime of the overclock working feels less responsive. It’s not that the overclock fail but that it feels like it becomes less responsive overtime (during the gaming session). I confirmed that the issue isn’t my computer as the system is still responsive when i switch to Kbm. My CPU is fine (temp usually 50-60C, usage at 50 percent is OC to 4.7Ghz i7900k. My ram is OC to its recommended 3200, at sits at 30 percent utilization. My Gpu isn’t OC but since I’m more competitive focused i usually have settings at low at its utilization sits below 30 percent. I bought a high quality mobo to support any system OC.  So thats not my issue. About a week ago now, I reinstalled windows and the first day of the OC the controller worked good (the whole session) next day back to how it was before the reinstall. Storage before and after reinstall had more than half of my C drive empty (1 terabyte SSD drive). Done normal window input latency adjustments to help, but the issue develops about 20-30 mins into my gaming session. The issue I have is the controller after some times becomes less responsive. Compared to the Default, the sticks are more responsive, but every other input is so much worse. Basically feels like I’m playing a much slower sense on a much higher ping. And on top of that, it feels like like the sluggishness / latency is inconsistent. Sometimes the controller is more responsive and other times its less. I don’t know whats happening but would like if you could give some insight into the problem. 

What I have: Scuf impact (PS4 variant) 
OC in software is 1000, but i checked with 500, the issue of sluggish-ness occurs just the same. 
What i know: this issue doesn’t affect FPS (240Hz capped ingame) or anything to do with system temps or processes. Usually monitoring to check for bad actors. 

So, if you would know whats causing this please let me know, it would be greatly appreciated.

LordOfMice, 2021-07-26:
https://github.com/LordOfMice/hidusbf/issues/78#issuecomment-877663607

> It’s not that the overclock fail but that it feels like it becomes less responsive overtime (during the gaming session).

"Feel" is bad term. Try to measure latency when you got this situation.

closed 2021-07-26 by LordOfMice

Christianguerra001, 2021-07-26:
Do you know of any programs that could measure latency? I use DS4Windows but it remains at 1ms even after the latency begins. So let me know if you know of any good programs.

bighammer25, 2022-03-24:
I’ve had this identical problem .. I just reinstalled windows did the oc worked great for about two hours.. I have the exact same controller I also play on a high end pc also my fps didn’t solve the problem either. If you ever found it out please let me know!!