LordOfMice/hidusbf #83: I really dont know what's the problem here

MasterDiggi, 2021-07-27:
https://user-images.githubusercontent.com/72347028/127084895-7f3c8163-8e18-4041-acc1-7afc0649c364.mp4

MasterDiggi, 2021-07-27:
I've tried moving it to the C:\Program Files and decrypting the files but it works the same.

LordOfMice, 2021-07-27:
1. Why do you think that this is problem?
2. You have to unzip contents before running setup, of course.

closed 2021-07-27 by LordOfMice

MasterDiggi, 2021-07-27:
> 1. Why do you think that this is problem?
> 2. You have to unzip contents before running setup, of course.

1 - Because it didn't do anything to the USB port neither the USB Controller
2 - I obviously unzipped the fine.

LordOfMice, 2021-07-27:
>Because it didn't do anything to the USB port neither the USB Controller

It's normal. Because "Filter On Device" do main job. You have to read readme.