LordOfMice/hidusbf #87: FPS Drops

Nxhyy, 2021-08-17:
I'm using a Dual Sense 5 controller but for some reason whenever I change the polling rate or put filter on it i seem to get a big decrease of fps by like 30 = 40 frames which is very big for me because my pc isn't the greatest. but i am running ds4 windows as well. if there is a fix for this issues could you let me know

LordOfMice, 2021-08-18:
No. This is problem with other software (probably not so good reaction to large number of input messages) and can not be fixed here.

closed 2021-08-18 by LordOfMice

Chaython, 2021-08-19:
Check your DPC latency, perhaps the high io from the usb is causing a wait.
Try setting MSIX for all devices, to hopefully help wait.

LordOfMice, 2021-08-19:
So high frame drops probably mean some problem in input part of game engine. Good engine has to be so input rate agnostic as possible. High polling rate itself has minimal impact on modern systems.

Nxhyy, 2021-08-19:
i checked my DPC latency and everything is on a low latency, i've also installed MSIX package but i don't know where to go from here.

Sinasharifi123, 2023-05-25:
I also have this problem can you please help me with your advices.