LordOfMice/hidusbf #91: Getting "Error trying to test loading of filter service: 193"

dmg007, 2021-08-28:
I'm using hidusb to overclock my gcc, but I keep running into this error. I reset windows, disabled secure boot, and disabled driver signature and it still doesn't work. I haven't had this problem on the other computer I downloaded this on. 

[image]

I'm pretty stumped and not sure of what else to do.

dmg007, 2021-08-28:
Read up on the other issues, seems like it isn't related to your software specifically. Still not working, but I might try reinstalling windows again.

closed 2021-08-28 by dmg007

LordOfMice, 2021-08-30:
>I might try reinstalling windows again

REinstalling may not help. If not then try clean install -> try hidusbf -> setup your usual software -> check that hidusbf is working still. And I am really interested to know what is the root of this problem.

LordOfMice, 2021-08-30:
https://github.com/LordOfMice/hidusbf/issues/54