LordOfMice/hidusbf #95: Possible old version

Rectonic2, 2021-10-01:
I was curious as to why my hidusbf.sys file has the date modified of 2016, while another persons has the date modified of 2021. Its just a question if i am in fact using something wrong. A response would be great thankyou.

[image]

Rectonic2, 2021-10-01:
I was also wondering if I should have anything running in services.msc as well.

LordOfMice, 2021-10-01:
>my hidusbf.sys file has the date modified of 2016, while another persons has the date modified of 2021

Because this is really different builds from different packages and different models of using.

>I was also wondering if I should have anything running in services.msc as well.

I think it's time to RTFM. There is described what this software does in system and even HOW. The third P.S. from https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/

closed 2021-10-01 by LordOfMice