LordOfMice/hidusbf #99: Cannot overclock

Rameezuwu, 2021-10-31:
Today my windows updated and my mouse went back to 125hz I tried to overclock it again but it didn't work...
I've been using this for months now and it worked great. Idk what to do so I am saying it here.

LordOfMice, 2021-10-31:
>Today my windows updated

I think you have to update this software too. I assume this is obvious.

closed 2021-10-31 by LordOfMice

Rameezuwu, 2021-10-31:
> > Today my windows updated
> 
> I think you have to update this software too. I assume this is obvious.

> > Today my windows updated
> 
> I think you have to update this software too. I assume this is obvious.

I am using the latest one.....

LordOfMice, 2021-10-31:
In this case there is clear recommendation in manual what you have to do. But if you read it, you would know that "the latest" is not a good term.

Rameezuwu, 2021-10-31:
> In this case there is clear recommendation in manual what you have to do. But if you read it, you would know that "the latest" is not a good term.

mmm I will redo this thing and see if it works thanks for your time

Rameezuwu, 2022-10-14:
> Today my windows updated and my mouse went back to 125hz I tried to overclock it again but it didn't work... I've been using this for months now and it worked great. Idk what to do so I am saying it here.

forgot to edit this but I fixed it