LordOfMice/hidusbf #102: Difficulties Uninstalling hidusbf

redline99cs, 2021-11-14:
Hello i want to uninstall this software because my mouse is lagging on one overclocked port and i accidently overclocked 2 ports and now im unable to revert back.
I did run the HIDUSBFU uninstaller but ds4windows is still reporting 1.5ms on my PS4 Controller (normal value should be 3.5ms).
The values in Setup.exe are all default. I tried searching system32 for the HIDUSBF.sys driver but i dont have it.
I dont know anymore how to revert back to old port speeds. 
If anyone can help it would be appreciated.

LordOfMice, 2021-11-15:
If you don't have hidusbf.sys in your system then this is something unrelated to this software. There is not possibility to influence on system that way without loading driver. And yes, the last resort as readme recommends is delete device in Device Manager and Reboot system.

closed 2021-11-15 by LordOfMice

EBGPFP, 2024-03-16:
> Hello i want to uninstall this software because my mouse is lagging on one overclocked port and i accidently overclocked 2 ports and now im unable to revert back. I did run the HIDUSBFU uninstaller [...]

I´m getting the exact same problem, did you manage to solve it?