LordOfMice/hidusbf #117: Fresh install from Win10 to Win11

ehbloody, 2022-01-12:
I've been using this program for years to OC my controller and hats off to @LordOfMice for creating it in the first place.  From what I understand nothing has changed within the program from Win10 to Win11 - however after doing a fresh install of Windows from 10 to 11, I can't apply an OC to my controller anymore. I read #116 and #100 but nobody seems to have a thorough answer as to what is causing this. My Secure Boot is disabled, Antivirus is all off, I installed it the exact same way as I did in the past but still nothing. Would really appreciate some assistance here if possible. Thank you. (please disregard the Red 'Generic Bluetooth Adapter' in the pictures as that is not the device I am trying to filter on)

[image]

[image]

ehbloody, 2022-01-12:
I somehow figured it out. I deleted my Secure Boot Keys in my BIOS. I have no idea how this would do anything but I am happy it did.

LordOfMice, 2022-01-12:
577 error is definitely problem with driver sign checking:
https://github.com/LordOfMice/hidusbf/issues/16

>My Secure Boot is disabled
No, it is enabled.
And it is easy to check:
msinfo32.exe ->

[image]

closed 2022-01-12 by LordOfMice

ehbloody, 2022-01-12:
Thanks Lord