LordOfMice/hidusbf #118: Bitlocker Recovery key code

aiyad11, 2022-01-13:
Hey there, I tried to disable the secure boot but I got stuck in the recovery key part I didn't know how to get it, any other solutions?.

closed 2022-01-13 by LordOfMice

aiyad11, 2022-01-14:
Hi, bro I read what you said and I tried my best but I didn't know how to do it can you help us please.

LordOfMice, 2022-01-15:
No, you didn't. Ask the RIGHT question.

aiyad11, 2022-01-17:
I really need help because I didn't know how to do it can you send us a link or a video to watch how to do and apply the steps.

LordOfMice, 2022-01-17:
Once again, until you using words "it","what","this" I continue to think that you didn't read readme because there is not SUCH words. There is some other, EXACT words describing what you have to do. Ask me WHAT you need to do (but don't know how) and I will answer.