LordOfMice/hidusbf #120: Device doesn't restart after 10 seconds error....

Shatter710, 2022-01-20:
Hi so sorry to bother you love your tool but I am having an issue...I am on windows 11 and I have secure boot disabled I verified in system info and I am also running setup as an admin it shows the rate change to 1000 and filter on device works doesn't give me the 577 error but when i try to restart after changing the rate it says device wont restart after 10 seconds and if i restart my pc it just goes back to default rate if i unplug and plug the controller back in it will change the blnterval to 1 instead of 4....i added both setup.exe and sx64.exe as exceptions in windows defender and i defiantly unzipped the program no luck any help would be greatly appreciated I attached a screen shot of the error hopefully it helps thanks!!!

[image]

LordOfMice, 2022-01-21:
>if i unplug and plug the controller back in it will change the blnterval to 1 instead of 4....i 

this means you overclock your device (hidusbf driver is loaded and functioning)

>if i restart my pc it just goes back to default rate

IDK what is the source of this problem. And I don't have PC with Windows 11 now so can not help.

Shatter710, 2022-01-21:
Sorry im really bad at trying explain stuff lol after restarting pc filter on device is unchecked and rate goes back to default and then blnterval goes back to 4 which is what it was from the start....and even when i unplug then plug the controller back in and the rate changes to 1000 i assume the overclock is applied but when i check with XinputTest my polling rate is still at 150 which is what it was out of the box before i tried to overclock....sorry for the confusion and thanks for the reply!! <3

LordOfMice, 2022-01-21:
>but when i check with XinputTest my polling rate is still at 150 which is what it was out of the box before i tried to overclock

This is the SECOND problem. I see on your screenshot that you have Asmedia xHCI controller and use it for your device - this can be the source. Connect device to USB port of AMD xHCI controller. AMD xHCI controller is far from ideal for overclocking too but you can try...

>after restarting pc filter on device is unchecked

This is definitely the problem with Windows 11 (or some option enabled in Windows 11 and disabled in Windows 10- by default) but as I said I can not help now. Try asking about this problem in forums dedicated for OS using, probably some Microsoft site is good point. This can be described in documentation or forums for driver developers too, but you have to ask RIGHT questions there.

Shatter710, 2022-01-21:
So I changed usb ports I had no idea that different ports had different controllers the asmedia or whatever....it now shows AMD xHCI but still no luck when overclocking I am starting to think its the controller it self because if i go any lower than a rate of 125 the controller will just start to turn it self off and on I am gonna try it on a different controller tomorrow like a stock xbox controller....its no worries I appreciate the help very much!!! <3 <3 I will try posting in forums dedicated for OS as you suggested hope you have a great day thanks for everything <3

Shatter710, 2022-01-21:
So I tried with a stock xbox controller today and got the same results as the other one....maybe its due to me being on windows 11 and like you said the AMD xHCI is not ideal for overclocking but unfortunately they are the only usb ports i have other than the 2 asmedia ones so I guess I am shit out of luck....still think you have done an amazing job with this program take care!

closed 2022-01-22 by Shatter710