LordOfMice/hidusbf #129: Don't work on windows 10 (ms 1.1a)

noregrets1, 2022-02-17:
It was worked perfectly for intellimouse 1.1a on windows 7 i installed windows 10 and it doesn't work. It doesn't show any error, it accept the HZ i choose, but in mouse freq tests just don't show higher than 125

LordOfMice, 2022-02-17:
You have to read readme (the link there)

closed 2022-02-17 by LordOfMice

noregrets1, 2022-02-17:
на 62hz ставит значит драйвер работает, тогда в чем причина? в ридми написано

 либо мышь не разгонябельна
  либо USBPORT.SYS/USBXHCI.SYS неоригинальный или слишком новый и поэтому
  неизвестен программе.

мышь разгонябельна так получается проблема в проге?

noregrets1, 2022-02-17:
версия драйвера контроллера совсем новая, 19.01.2022 в диспетчере устройств

noregrets1, 2022-02-18:
дата разработки 19.01.2022 версия 10.0.19041.1503 такая же, драйвер от майкрософт
не работает все равно,

LordOfMice, 2022-02-21:
Надо внимательно прочитать линк, там НАПИСАНО в чём особенность разгона под 10 x64. Незнание английского не освобождает от чтения, сейчас есть онлайновые переводчики.

noregrets1, 2022-02-21:
Друг, я все читал и все делал. Вообще я как пользователь должен просто включить прогу  выбрать герцовку и закрыть прогу, все. Так оно и работало на вин 7, никаких вопросов.  Если оно здесь не работает, то оно не работает, все просто.

LordOfMice, 2022-02-22:
>Друг, я все читал и все делал.
А я вижу прямо противоположное. Причём мне очень просто это понять по вопросам.

>Вообще я как пользователь должен просто включить прогу выбрать герцовку и закрыть прогу, все.

https://github.com/LordOfMice/hidusbf/issues/16#issuecomment-883918974
Жду предложений.

noregrets1, 2022-02-22:
Там написано черным по белому, если герцовка понижается значит драйвер работает нормально. У меня прога ставит ниже 125 гц, 62 и 31. Выше никак не работает, значит драйвер работает нормально. Мышь разгонябельна, драйвера официальные. Так в чем же проблема? Может в разработчике програмы? Ведь дальше там написано, что програма может не видеть новых версий драйверов, и поэтому может не работать.

  If the rate does not increase more than 125Hz, then you should
  try DECREASE it to 31Hz or 62Hz.
  If the rate does decrease, then the driver functions properly, however
  or you mouse is not overclockable; or USBPORT.SYS/USBXHCI.SYS was altered
  or simply too new and so unknown to the program.

LordOfMice, 2022-02-23:
Я третий и последний раз предложу прочитать информацию по линку в ридми, он является НЕОТЕМЛЕМОЙ частью ридми и даже уже приведён мной в этом обсуждении. Там информация на английском, но незнание английского НЕ ЯВЛЯЕТСЯ оправданием её не читать.

noregrets1, 2022-02-23:
я в 10 раз повторяю там нет того что помогло бы мне заставить работать программу, я 1 раз прочитал все. Ты бы поубавил свое ЧСВ и нормально помог или обновил программу.