LordOfMice/hidusbf #132: Controller OCed perfectly, but after restart it stopped working

Manic137, 2022-03-09:
I'm using a DS4 controller, although I doubt that matters. I restarted my PC to turn off the secure boot, then overclocked it successfully. I then restarted my pc again to turn off the secure boot as Apex Legends Anti Cheat requires that you have secure boot on(game would not launch with it disabled). This however caused the OC to malfunction and not work(controller wasn't usable, red bar in hidusbf). Thankfully I did not have to restart the PC to disable the OC and the controller worked again. I read there's a way to only allow hidusbf through the registry and not affect Apex Legends?

LordOfMice, 2022-03-10:
>Apex Legends Anti Cheat requires that you have secure boot on
>I read there's a way to only allow hidusbf through the registry and not affect Apex Legends?

IDK does it help with your game (Anti Cheat engines can verify many things) but correct, it is possible to have Secure Boot On with hidusbf loaded and i personally checked this on Windows 10 x64 version 20H1 last time. 
You have to read this:
https://www.geoffchappell.com/notes/security/whqlsettings/index.htm
(WhqlSettings).
P.S. Looks like you are the first one who carefully read readme ;) This is the first right question due to 3 year.

closed 2022-03-10 by LordOfMice

Manic137, 2022-03-10:
> IDK does it help with your game (Anti Cheat engines can verify many things) but correct, it is possible to have Secure Boot On with hidusbf loaded and i personally checked this on Windows 10 x64 [...]

This did the trick for me, worked so far, I'll update if anything changes :D
[image]

> P.S. Looks like you are the first one who carefully read readme ;) This is the first right question due to 3 year.

Geez xD

rstewa35, 2022-03-12:
> This did the trick for me, worked so far, I'll update if anything changes :D [image]

I can also confirm this works on Windows 11 Pro x64

[image]

LordOfMice, 2022-03-13:
>I can also confirm this works on Windows 11 Pro x64

Just for completeness - try to check other identical option too:
[HKEY_LOCAL_MACHINE\System\CurrentControlSet\Control\CI\Policy]
WhqlSettings:DWORD=0x00000001

First disable(delete) UpgradedSystem, of course.

rstewa35, 2022-03-13:
> Just for completeness - try to check other identical option too: [HKEY_LOCAL_MACHINE\System\CurrentControlSet\Control\CI\Policy] WhqlSettings:DWORD=0x00000001
> 
> First disable(delete) UpgradedSystem, of course.

That key appears to work too

[image]

LordOfMice, 2022-03-13:
Ok, thank for checking. So there is no difference of Windows 10 and 11 right now.

SergioVB-99, 2022-03-18:
[image]

SergioVB-99, 2022-03-18:
Ok, this works fine in Windows 11 !!!!   Thanks
[image]

n00batt, 2022-04-15:
How come when I look in my command prompt, there is no whql settings, as well as none when i check in my registry editor? Do I need to do something in order to be able to see this.

LordOfMice, 2022-04-16:
>in my registry editor

...just create new value.

n00batt, 2022-04-16:
Sorry haha don't do this often. Managed to get it working using WhqlSettings key on windows 11.

OsakaRider, 2022-04-18:
both UpgradedSystem and WhqlSettings work fine for me Windows 11 fresh install , but dont forget you need to restart after adding the reg key.

sleeps-weeps, 2022-07-21:
How do I bypass secure boot I don't get the valve WhqlSettings REG_DWORD  0x1 HOW do I do that?

LordOfMice, 2022-07-22:
Answer is few posts above.

sleeps-weeps, 2022-07-26:
Okay thank you got it to work

St1llest, 2022-07-26:
I'm confused. I don't know how to use the command console. How exactly should I be inputting this in the command console?

LordOfMice, 2022-07-27:
Do you know what is "registry editor"? It is *GUI* application.

St1llest, 2022-08-04:
> Do you know what is "registry editor"? It is _GUI_ application.

I'm learning so I know what it is, but I don't know how to use it yet. How would I set this up so that I can bypass with secure boot on?

LordOfMice, 2022-08-05:
> I don't know how to use it yet

This question was already answered in this issue. In any case, I assume you have to use search and read something else rather than asking unrelated to THIS software questions.

rstewa35, 2022-09-08:
I created a guide on how to do this to help those unfamiliar with the registry. You can find it [here](https://gist.github.com/rstewa35/579e6003aed35572cdecfcc888ceefd6).

emilovz, 2022-12-14:
[image]

it works but not with my wmo, any idea?

LordOfMice, 2022-12-14:
>any idea?

1. You have to CAREFULLY read what you see on the screenshot.
2. And there is some addition:
https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/post-28962318

emilovz, 2022-12-15:
So I need the latest hidusbf which is in hidusbfn, available in the hidusbff "signed" package, which is impossible to use without atsiv/umap or test mode? I need 500hz overclock without touching atsiv,umap,test mode, secure mode on w10 22h2, that's impossible atm?

Zts789, 2023-04-22:
> I created a guide on how to do this to help those unfamiliar with the registry. You can find it [here](https://gist.github.com/rstewa35/579e6003aed35572cdecfcc888ceefd6).

I know this is so late in the future but if you still have the guide can you make a new working link? thanks ahead

St1llest, 2023-04-22:
Got you. Use this video. Shows it all.

https://youtu.be/my_JsBRAFsQ

rstewa, 2023-05-02:
> > I created a guide on how to do this to help those unfamiliar with the registry. You can find it [here](https://gist.github.com/rstewa35/579e6003aed35572cdecfcc888ceefd6).
> 
> I know this is so late in the future but if you still have the guide can you make a new working link? thanks ahead

I updated the link in my original comment but I'll also add it here. You can find my written guide [here](https://gist.github.com/rstewa/579e6003aed35572cdecfcc888ceefd6)

eriiicccc, 2023-05-31:
[``](
[image]
)

eriiicccc, 2023-05-31:
[image]

Porknpig, 2023-06-16:
@LordOfMice  I appreciate your response but this doesn’t make any sense to me. Maybe i’m dumb but what i’m getting from this is how to enter apex legends without effecting hidusbf through the directory by enabling secure boot? I’m asking how to fix my b interval being stuck on 4 without any applications running. on 1000hz filter checked it reads binterval 4 and on 8000hz it reads binterval 1. I’ve reset and watched as many youtube videos one can watch and nothing. Could you possibly explain what I should be grabbing from issue 132?

Porknpig, 2023-06-16:
@LordOfMice does this mean my secure boot is causing the binterval of 4? And if I change directory for it to work on windows 10 it’ll work properly and go to binterval of 1?

rafadorock4, 2023-09-27:
This .reg file worked flawlessly in Win11-23H2 .

`Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\CI\Policy]
"UpgradedSystem"=dword:00000000`

For those that need it, put these 3 lines above in notepad and save as .reg then open it, click yes and reboot the system.

Adeystix, 2023-11-24:
i've tried everything including whql settings and upgraded system i still keep getting error 193 , the only way i've gotten it to work on windows 11 is by disabling memory integrity but anticheat software doesn't allow me play with integrity turned off

Usufferance, 2023-11-30:
[image]
[image]
[image]
[image]

I'm Sorry I don't understand what is wrong. How do I enable  Memory Integrity. (not good with tech)

Adeystix, 2023-11-30:
How do I enable Memory Integrity. (not good with tech)

you have to turn it off from core isolation page in windows security settings . looking for a way to get it to work without turning that setting off

LordOfMice, 2023-12-02:
>How do I enable Memory Integrity.

I assume the answer is obvious - uninstall this software as described in readme.

> looking for a way to get it to work without turning that setting off

wiki: how could I help to make this software better: https://github.com/LordOfMice/hidusbf/issues/16#issuecomment-883918974

no-identd, 2024-02-25:
https://winraid.level1techs.com/t/tips-discussion-usage-of-mod-signed-drivers/31158

Theoretically installing the cert in the manner outlined above should obviate needing either test signing or registry hacks

OsakaRider, 2024-02-26:
> https://winraid.level1techs.com/t/tips-discussion-usage-of-mod-signed-drivers/31158
> 
> Theoretically installing the cert in the manner outlined above should obviate needing either test signing or registry hacks. Haven't yet tested it tho

I remember testing this before a while ago and it does not work, not to mention this is an old method if it worked this thread would be in need to begin with

no-identd, 2024-02-27:
> I remember testing this before a while ago and it does not work, not to mention this is an old method if it worked this thread would be in need to begin with

All methods I've seen described so far as they relate to usbhidf don't appear to place the certificates into the correct certificate storeS. Take a close look at the Win-raid install scripts. Perhaps it won't work — I had intended to test this yesterday, but unfortunately I now won't be able to do so until in quite a few days, but I WILL test it