LordOfMice/hidusbf #136: Controller on pc

bighammer25, 2022-03-25:
I am having the identical problems as this person was having here link 

https://github.com/LordOfMice/hidusbf/issues/82

i have to reinstall windows and it works for a few hours then becomes slow and sluggish I used ds4 to monitor input and it goes to 0.5 ms controller  reacts fast but it’s all buggy.

LordOfMice, 2022-03-25:
Once again, it is not the problem of this software, so i can not help.
And yes, I know few cases when instability after overclocking was caused by bad USB cable of device, so you can check this.

>it works for a few hours then becomes slow and sluggish

You can read some of the "latency optimization" manuals like this:
https://www.overclock.net/threads/usb-polling-precision.1550666/
or this:
https://github.com/BoringBoredom/PC-Optimization-Hub

And for isolation of problem you can try to check different OS. Win 7 is known example of stable results in area of USB input.

closed 2022-03-25 by LordOfMice

bighammer25, 2022-03-25:
I’ll be honest I figured it wasn’t but you would be the man who would have the answers to this problem and I was correct lol thanks! I’m gonna look into I’m just trying to play some cod at a high level with out this input lag. Lol

Annika1986, 2023-07-04:
@bighammer25  still no help.😔 do you have a solution now?

sepyum22, 2024-08-17:
Im the same exact situation.. no fix ??

Annika1986, 2024-08-17:
Nope sorry Gesendet mit der mobilen Mail AppAm 17.08.24 um 14:10 schrieb sepyum22
                
            
            
                
                    Von: "sepyum22" ***@***.***>Datum: 17. August 2024An: "LordOfMice/hidusbf" ***@***.***>Cc: "Annika1986" ***@***.***>,"Comment" ***@***.***>Betreff: Re: [LordOfMice/hidusbf] Controller on pc (Issue #136)
                
                
Im the same exact situation.. no fix ??

Annika1986, 2024-08-17:
I think it’s a game issues. In my opinion Fortnite Gesendet mit der mobilen Mail AppAm 17.08.24 um 14:10 schrieb sepyum22
                
            
            
                
                    Von: "sepyum22" ***@***.***>Datum: 17. August 2024An: "LordOfMice/hidusbf" ***@***.***>Cc: "Annika1986" ***@***.***>,"Comment" ***@***.***>Betreff: Re: [LordOfMice/hidusbf] Controller on pc (Issue #136)
                
                
Im the same exact situation.. no fix ??