LordOfMice/hidusbf #142: XPSP3, Logitech USB RX250 don't overclocked

z0hm, 2022-04-16:
Device:
"USB\VID_046D&PID_C050\5&385A3465&0&2"
"USB\Vid_046d&Pid_c050&Rev_2720"
"USB\Vid_046d&Pid_c050"
  Name: Logitech USB (RX250) Optical Mouse
  Driver: HidUsb
  Binary: system32\DRIVERS\hidusb.sys
    Path: C:\WINDOWS\system32\DRIVERS\hidusb.sys
    Version: 5.1.2600.5512 (xpsp.080413-2108)
  Binary: HIDUSBF.SYS
    Path: C:\WINDOWS\System32\DRIVERS\HIDUSBF.SYS
    Version: 1.2.0.0
  LowerFilters: hidusbf,LUsbFilt
  BusSpeed: 0
  bInterval: 2
  Overrided.bInterval: 2

Controller:
"PCI\VEN_8086&DEV_3A34&SUBSYS_82D41043&REV_00\3&11583659&0&E8"
"PCI\VEN_8086&DEV_3A34&SUBSYS_82D41043&REV_00"
"PCI\VEN_8086&DEV_3A34&SUBSYS_82D41043"
"PCI\VEN_8086&DEV_3A34&CC_0C0300"
"PCI\VEN_8086&DEV_3A34&CC_0C03"
  Name: Intel(R) ICH10 Family USB Universal Host Controller - 3A34
  Driver: usbuhci
  Binary: system32\DRIVERS\usbuhci.sys
    Path: C:\WINDOWS\system32\DRIVERS\usbuhci.sys
    Version: 5.1.2600.5512 (xpsp.080413-2108)
  Binary: USBPORT.SYS
    Path: C:\WINDOWS\System32\DRIVERS\USBPORT.SYS
    Version: 5.2.3790.3959 (srv03_sp2_rtm.070216-1710)

Is the problem in the hardware or software?

z0hm, 2022-04-16:
Binary: USBPORT.SYS
    Path: C:\WINDOWS\System32\DRIVERS\USBPORT.SYS
    Version: 5.1.2600.6437 (xpsp_sp3_qfe.130809-0417)

Also 125Hz only

LordOfMice, 2022-04-16:
>Is the problem in the hardware or software?

Hardware. To be precise - this mouse.

closed 2022-04-16 by LordOfMice