LordOfMice/hidusbf #145: HID-compliant Controller Turns Orange...

ShribbleDibble, 2022-04-22:
I have referred to a tutorial video to overclock my ps4 controller using your program, @LordOfMice, but instead of the HID-compliant controller changing its B-interval to 1, the entire row becomes orange. Any suggestions?

Kind regards,
Cody.

LordOfMice, 2022-04-22:
There is warning in readme and two pinned issues about this.

closed 2022-04-22 by LordOfMice

LordOfMice, 2022-04-23:
!!! Warning 2 - you SHALL disable Secure Boot or modify the Registry accordingly to enable loading of Microsoft unsigned drivers to load HIDUSBF driver successfully under Windows 10 x64 version 1607 (Anniversary Update) or newer versions.

ShribbleDibble, 2022-04-23:
Thank you. One problem, I don't know what that means. I'm just an ordinary consumer who wants lower input lag for Rocket League. Could you kindly explain how I disable Secure boot or modify the Registry?

Regards,
Cody.