LordOfMice/hidusbf #156: 4 bInterval on 1000hz polling rate

user-wizard, 2022-07-17:
[image]
I have a PS5 controller and on Windows 10 at 1000hz there was 1 bInterval, but after overclocking on Windows 11, bInterval changed. XInput test shows still shows 1000hz, but I am just curious about why this happened

LordOfMice, 2022-07-17:
[Copy IDs] has enough information for the first point of problem isolation.

Dama624, 2022-07-17:
I'm not the thread creator but I'm running into the same issue as well; here is the output of `Copy IDs` after I installed on my controller:
```
Device:
"USB\VID_054C&PID_0CE6\6&3AF0F9CE&0&7"
"USB\VID_054C&PID_0CE6&REV_0100"
"USB\VID_054C&PID_0CE6"
  Name: USB Composite Device
  Driver: usbccgp
  Binary: \SystemRoot\System32\drivers\usbccgp.sys
    Path: C:\WINDOWS\System32\drivers\usbccgp.sys
    Version: 10.0.19041.488 (WinBuild.160101.0800)
  Binary: HIDUSBF.SYS
    Path: C:\WINDOWS\System32\DRIVERS\HIDUSBF.SYS
    Version: 1.2.0.0
  LowerFilters: hidusbf
  BusSpeed: 2 (High)
  bInterval: 4
  Overrided.bInterval: 4

Controller:
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21&REV_01\4&2C18E2E3&0&000B"
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21&REV_01"
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21"
"PCI\VEN_1022&DEV_43D5&CC_0C0330"
"PCI\VEN_1022&DEV_43D5&CC_0C03"
  Name: USB xHCI Compliant Host Controller - AMD USB 3.10 eXtensible Host Controller - 1.10 (Microsoft)
  Driver: USBXHCI
  Binary: \SystemRoot\System32\drivers\USBXHCI.SYS
    Path: C:\WINDOWS\System32\drivers\USBXHCI.SYS
    Version: 10.0.19041.1566 (WinBuild.160101.0800)

OS: NT x64 (amd64)
```

rrazed, 2022-07-18:
same thing is happening to me, plus now it gives me an option for 8000 polling rate and it didnt used to do that either

LowkeyIzzy, 2022-07-18:
I just downloaded today and selecting 1000hz gave me a 4 Blnterval. Selecting 8000 gave me a 1 Blnterval. My Blnterval went down 1 each time I increased my HZ past 1000.

LordOfMice, 2022-07-18:
>BusSpeed: 2 (High)
+
https://github.com/LordOfMice/hidusbf/commit/624353ed879bd233487021ff4bb807af1c09140d

closed 2022-07-18 by LordOfMice

Dama624, 2022-07-18:
I downloaded the latest, redid the process, and it still seems to be occurring...
Here's the output of `Copy IDs`:
```
Device:
"USB\VID_054C&PID_0CE6\6&3AF0F9CE&0&7"
"USB\VID_054C&PID_0CE6&REV_0100"
"USB\VID_054C&PID_0CE6"
  Name: USB Composite Device
  Driver: usbccgp
  Binary: \SystemRoot\System32\drivers\usbccgp.sys
    Path: C:\WINDOWS\System32\drivers\usbccgp.sys
    Version: 10.0.19041.488 (WinBuild.160101.0800)
  Binary: HIDUSBF.SYS
    Path: C:\WINDOWS\System32\DRIVERS\HIDUSBF.SYS
    Version: 1.2.0.0
  LowerFilters: hidusbf
  BusSpeed: 2 (High)
  bInterval: 4
  Overrided.bInterval: 4

Controller:
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21&REV_01\4&2C18E2E3&0&000B"
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21&REV_01"
"PCI\VEN_1022&DEV_43D5&SUBSYS_11421B21"
"PCI\VEN_1022&DEV_43D5&CC_0C0330"
"PCI\VEN_1022&DEV_43D5&CC_0C03"
  Name: USB xHCI Compliant Host Controller - AMD USB 3.10 eXtensible Host Controller - 1.10 (Microsoft)
  Driver: USBXHCI
  Binary: \SystemRoot\System32\drivers\USBXHCI.SYS
    Path: C:\WINDOWS\System32\drivers\USBXHCI.SYS
    Version: 10.0.19041.1566 (WinBuild.160101.0800)

OS: NT x64 (amd64)

```

LordOfMice, 2022-07-18:
> it still seems to be occurring...
Occurring what? bInterval 4 _IS_ 1000Hz for HIGH SPEED devices...

Dama624, 2022-07-18:
> > it still seems to be occurring...
> 
> Occurring what? bInterval 4 _IS_ 1000Hz for HIGH SPEED devices...

ah, my bad; I was under the assumption that it was supposed to be at 1
thank you!

shwetunlin, 2023-10-29:
My bInterval is a 1 with 1000hz.