LordOfMice/hidusbf #160: Program Fried my Ports

MegashiNYC, 2022-07-21:
I tried your program to overclock my mouse and it doesn't feel 1to1 anymore, its like my computer isn't reading the full polling rate of my mouse I tried 4 different mouses and I cant control recoil for nothing mouses are not reading my small movements in game, is there anyway to fully revert everything or do I need a new motherboard... I have a Ryzen System btw when i use a polling rate checker my small movements are like 12hrtz - 100hrtz please help

LordOfMice, 2022-07-22:
>my small movements are like 12hrtz - 100hrtz
And what did you expect from _small_ movements?

closed 2022-07-22 by LordOfMice

MegashiNYC, 2022-07-22:
That's all you can reply to seriously? So you make a program and when it fails and messes up my mouse movements you cant even try to help me? I wasn't even being rude to you nor am I mad at you I just wanted some help but whatever ill buy a new motherboard and just replace the old one thanks again, brother.

LordOfMice, 2022-07-22:
Probably you misunderstand what this software really does. After you un-link driver from your device and uninstall driver it just doesn't have any chance to influence on your system. When i answered above i thought that you described functioning of your device WITH driver, not without. So you have some other problem. The algorithm to isolate such problem never changes - check this mouse on other system and check other mouse on this system.

MegashiNYC, 2022-07-22:
Alright.