LordOfMice/hidusbf #161: Info

GisoDsntMiss, 2022-07-22:
With the latest update i heard it's possible to overclock controllers to 8khz like it is possible for the zaunkoenig?

LordOfMice, 2022-07-22:
Where did you find this statement? "high speed devices support" is just correct information about such devices in setup, not the new function.

closed 2022-07-22 by LordOfMice

GisoDsntMiss, 2022-07-22:
On some discord, they said they were able to overclock controllers to 8khz and i was just curious about it

LordOfMice, 2022-07-22:
Once again. Driver itself has possibility to overclock ANY device literally from the first release day. But value you feed to driver depends from setup and setup did not have code to correctly process high speed devices.

>they were able to overclock controllers to 8khz and i was just curious about it

2k-8k overclocking is unrelated to the last changes in setup. And just follow the link in readme if you wish to know how to do this.

GisoDsntMiss, 2022-07-22:
So basically following the readme instructions for 2k/8khz you can overclock any device?

LordOfMice, 2022-07-22:
Yes. But whether you will get something useful depends on device as with any overclocking.