LordOfMice/hidusbf #162: Is it possible to overclock a controller connected with the Cronus zen ?

vxynnn, 2022-07-28:
hi, I wanted to ask if it was possible to overclock a controller connected via Cronus zen I tried but the imput delay is always the same

LordOfMice, 2022-07-29:
>Cronus zen

IDK what is that. This software works with any USB devices. Аnd if device is visible in Setup (with Interrupts or USB or All filters) then you can try to overclock it.

>but the imput delay is always the same

You should read about overclocking to higher USB speed class. Setup now warns for such case for Low Speed to Full Speed overclocking.

closed 2022-07-29 by LordOfMice

vxynnn, 2022-07-29:
I tried to connect it via this device but it gives me 10ms of delay with the ps5 controller 
if instead I connect it normally 0 ms

vxynnn, 2022-07-29:
the device in question and the Cronus zen serves to improve the aim and other things so

LordOfMice, 2022-07-29:
>via this device
This is processing delay. It's normal as processing has some time to be executed and can not fixed outside this device.