LordOfMice/hidusbf #170: hidusbf automatically resets to normal after ~5mins

MmmYogurt, 2022-08-23:
Hello! I've tried using hidusbf for a long time now and every time I use it, this issue keeps happening. So basically, whatever polling rate I set it to, after ~5mins, it resets back to the initial polling rate (even though the program doesn't show it changed, i.e. Filtered, 1000hz, 1bInterval). I've tested it on two different PS4 controllers, also on Windows 10 and Windows 11 and on different USB 3.0/2.0 ports. If you need any other info, ask away!

LordOfMice, 2022-08-24:
>it resets back to the initial polling rate

1. How did you check this?
2. And what will happen after "Reset" in setup in such situation?

MmmYogurt, 2022-08-24:
1. I'm playing Rocket League and I've been struggling with input lag in that game for a long time, so I can sense very easily every input lag changes. 
2. The initial input lag comes back, but the setup shows no changes or errors. For example, I follow the instructions on how to change the polling rate, I change it to 1000hz and everything goes well, no errors or anything else. Then I play the game and I can feel a big improvement in input lag and then, after ~5mins, I can feel how it randomly goes back to the initial input lag (probably 125hz), but nothing changes in the program, everything stays the same (it shows Filter on Device is ticked, 1000hz and 1bInterval) and also, I don't get any errors or anything else. (I attached a screenshot of the program after it "Resets")
[image]

LordOfMice, 2022-08-24:
>I can sense

I am not psychologist and will not help with "sense", "feeling" or what else. "Check" means "measure".

MmmYogurt, 2022-08-24:
Okay, so I've just tested and measured the polling rate, refresh rate and ms of input delay BEFORE using hidusbf, WHILE using hidusbf and AFTER it "resets", by using DS4Windows to emulate an Xbox controller (because I only have PS4 controllers) so I can then test the polling rate with a program that only works for Xbox controllers called "XInputTest". 
The results:
Before overclocking: Average of 4.95 ms
While overclocked: Average of 1.10 ms
After it "resets": Average of 5.38 ms

MmmYogurt, 2022-08-24:
Also, I checked this on both PS4 controllers with different cables and got the same ~average~ results.

LordOfMice, 2022-08-24:
Ok, probably it is some hardware problem. The next test is:
1. waiting "Reset" fact
2. go to Setup -> [Restart]
3. checking what happens after restart

If this will revert rate to overclocked value (1 ms) then the next test is check on different PC (as you tested two devices already; they CAN have the same problem but more probably problem is in PC hardware rather in device hardware).

LordOfMice, 2022-08-24:
P.S. And good test is try to overclock some other overclockable device on your system, like USB mouse, of course.

MmmYogurt, 2022-08-24:
It goes back to 1ms after restarting, but then again to 5ms after ~5mins. I've tested it on my roommates PC and it seems like it runs with no problems with both my controllers. I've also tested my mouse and the same thing happens as with my controllers. Do you know what hardware issue could make this happen?

LordOfMice, 2022-08-24:
What USB controller are you using on problem PC? Asmedia (and thus AMD) is known for problems.

MmmYogurt, 2022-08-24:
Yes, I've checked on two OS, and I've also reinstalled Windows 10 a few times recently and the same issue occurs.
Well then, Thank you a lot for your time! I will just quit playing Rocket League in this case LOL. Thanks again for your assistance!

LordOfMice, 2022-08-24:
And the "last hope" advice - try to connect device through USB hub...

closed 2022-08-24 by LordOfMice

Noneed2explain, 2022-09-01:
bruh i just wrote about the same issue and didnt know i wasnt alone with it. Idk if its rocket league itself or not but now im confused af /:

LordOfMice, 2022-09-01:
The only thing you can try now is the last one - it is not tested still (or no result published).

MmmYogurt, 2022-09-01:
Mine wasn't connected through a USB hub.
*Haven't tested the other way so I guess it's worth testing.

LordOfMice, 2022-09-02:
And yes, the only workaround i can suggest is to restart device periodically on timer but it is very rude solution.

Noneed2explain, 2023-01-06:
@MmmYogurt Did u solve ur Problem? If so, what did u do to fix it? Would be sick if could share ur Solution.

MmmYogurt, 2023-05-09:
> @MmmYogurt Did u solve ur Problem? If so, what did u do to fix it? Would be sick if could share ur Solution.

Sorry, I didn't solve it

> but more probably problem is in PC hardware rather in device hardware). 

@LordOfMice I have recently upgraded my whole PC, meaning I built it from scratch and I'm on AMD now, still the same problem occurs. So it doesn't seem to be a PC hardware problem then. Tested with several other devices (controllers, wired/wireless mice and keyborads) still same issue occurs. Would you have other suggestions?

LordOfMice, 2023-05-10:
>I'm on AMD now
>Tested with several other devices (controllers, wired/wireless mice and keyborads) still same issue occurs.

With such symptoms it is some software you are using.
Something like this: https://github.com/LordOfMice/hidusbf/issues/201

Annika1986, 2023-07-04:
@LordOfMice i have the same problem. have you found a solution in the meantime?

Annika1986, 2023-07-04:
@MmmYogurt have you found a solution in the meantime ? ? 😔

Annika1986, 2023-07-04:
@LordOfMice i don‘t use gforce experience. so the problem cannot be here😔

TheGr8t, 2024-01-10:
Hi, I'm Also getting this issue.

I been playing this game at a higher rank for a while, understanding precise movement for years now. I understand all times of input delay issues within a competitive mindset, having a bad day, or slump, or just being bad and can't control the car. When i play ranked at a higher level i need to jump or turn at exact moments to get prefect control or normally easy saves for me, I mastered every difficult thing in the game beating pros winning tournament's etc.

however, I'm also getting a input delay issue that i ''feel'' I don't know what other word I can use, but i know for a fact is that there is something going on with this software not working after a few things.

I had this plugin for a while on ps5 controller and everything, at some point my controller got stick drift like a normal controller after a while. I got the Dual sense egde controller and overclocked it, BAM thats when my issue started, The software had 100% noticeable input delay it never got worse or better, for months now. I even went back to old wire and controller and still felt the same issue.

Idk if i can list everything but ill try ( been attempting a fix for MONTHS now)
Deteled/Reseting Windows MORE THEN ONCE
1st i reinstall bluetooth 
Updated controller drivers
Deleted controller drivers
Turn low latency to ultra
Upgrade drivers
Rollbacking drivers
Update windows
Delete the overclocking system. 
Reinstall steam epic games and rocketleague x 2
Change extra settings within game files of rocket league
Used xbox controller
I attempted playing another game but i couldn't tell cause idk what its like to have no delay on that game cause i only play rocket league 
Updating motherboard (it felt better for like a few hours, but it was all in my head ig)
Delete services that i don't use
Deleted all services but Microsoft ( it did make the sens much higher, which made more sense on why pros play with 1 or 2 sens it was very very loose, however it made me realized that the 1st input was Deleted) 
Ds4 polling

Honestly i forgot a few i think, but that covers most, i think.

I got a new pc setup with a new controller and wire for the controller and monitor and i still have input delay not as much as the old setup.

What else would i be able to do to get it working for the new setup or the old setup? Has anyone else been able to fix this problem? Is this for just rocket league? It worked before but i still can feel the delay

Noneed2explain, 2024-01-10:
hey @TheGr8t can u give me your discord? Idk how to dm u here

Annika1986, 2024-01-10:
@TheGr8t  sorry, I think I have to disappoint you. It's not because of this program. I now also have it on the PS5 and the OC program doesn't work there at all. I have it worst with the edge. least of all with the Astro but it will probably never go away again. also had a test PC here. (same problem) I just don't know what it is and will probably never find out. I did everything you did and much more. But I went from being an extremely good player to a medium to worse player just because of this stupid problem. I think it’s epic.  sorry also for my grammar, I come from Germany.

Noneed2explain, 2024-01-10:
hey @Annika1986 im also from germany, can u give me your discord?

Noneed2explain, 2024-01-10:
i have some questions to both of u and i think its better to talk so we are on the same page since my english is bad and i dont want to write my a$ off

Annika1986, 2024-01-10:
@Noneed2explain  ich denke nicht das ich dir irgendwie weiterhelfen kann. 😔 hab das Problem ja selbst heftig. Mein DC kann ich hier nicht öffentlich schreiben, da es auch mein gamertac ist. Worum geht es denn genau ?

TheGr8t, 2024-01-10:
My discord is great0515

TheGr8t, 2024-01-10:
I just reset my usb ports on my old pc, didn't fix the issue however I did notice that its more sensitive then. before.
I litterly just played like over 10 rank games last night, and just ran  rank for the past hour after resting port redoing the overclock on both 1000 and 8000

1000 is better in my opinion
but 8000 seems more stable is that makes sense

TheGr8t, 2024-01-10:
im sure there is a fix, i just havent seen or figured out yet