LordOfMice/hidusbf #176: Ps5 Controller not actually overclocking

Moklakins, 2022-09-08:
Hello,

I'm having an issue trying to overclock my PS5 controller.  After I attempt to overclock the controller it shows yes under filter, rate 8000, and bInterval 1, but when I test it, the tests show a max polling rate around 500hz.  I am testing it by using steam input to make the right thumbstick a mouse then using multiple mouse polling rate checkers, and they all show the same results.  I overclocked my PS4 controller the same way (except rate is 1000), and using the same methods to test, it is correctly showing a max polling rate around 1000hz.

I also tried using lower rates for the PS5 controller and no matter what I have it set to it stays at 500hz.  Anyone have any tips to try?

rstewa35, 2022-09-08:
I checked my duelsense controller with [DS4Windows](https://github.com/Ryochan7/DS4Windows) and https://devicetests.com/mouse-rate-test . I was getting 1000hz fine when I had it set to 1000. I also got 1600hz when it was set to 8000. Maybe Steam input limits the polling rate.

Moklakins, 2022-09-09:
I would think that if steam input was limiting polling that the ps4 controller would also be rate limited, but mine tested at 1000hz using steam input.  It is still possible though as steam input does differentiate between a ps4 and ps5 controller. I'll test using ds4 Windows next time I'm on my PC.

Moklakins, 2022-09-09:
Sorry typed this on my phone and accidentally closed the issue.

Moklakins, 2022-09-09:
You were right, I just exited steam and tested with ds4windows and I am now getting the same results as you.  Time to send a ticket to steam.

closed 2022-09-09 by Moklakins