LordOfMice/hidusbf #181: Hidusbf overclocking error

Nuke1216, 2022-10-24:
I'm receiving two problems where I download the Hidusbf for overclocking my controller and after i "extract all" I get this first pop up that's asking me to download an app from the app store when I click on setup, but no app is available. Then my second problem is when I download the app and don't extract the driver file, I'm able to access the setup to continue the process to overclocking controller but I'm lead with another problem that says " Filter driver Hidusbf is disabled or not installed" if i click continue setup my controller line becomes a red line. Please help I am on windows 11 if that matters 
[image]
[image]

LordOfMice, 2022-10-25:
It's time to read readme and wiki. You have to extract .zip content and install hidusbf service before anything else. But it is NOT complete list you have to do on Windows 11.

closed 2022-10-25 by LordOfMice

Nuke1216, 2022-10-25:
Hey is there anyway I could contact you via phone to have you walk me
through this Im not understanding the read me section honeatly I’m so
confused when it comes to pc things

LordOfMice, 2022-10-26:
No. And there are enough manuals (texts and videos). Just google "hidusbf".