LordOfMice/hidusbf #188: Windows version too recent?

Souillon, 2022-11-23:
Hello,

I was using this software for my scuff controller on my old computer and on my new one it doesn't work anymore! 
Is this due to my 5600h processor or to my version of Windows 22h2 which is not yet compatible? 

Thank you for your feedback and thank you for your useful work! @LordOfMice

LordOfMice, 2022-11-24:
>Windows 22h2

IDK what is that.

daflamingpotato, 2022-11-26:
> 

go to windows search and type "winver", 22H2 is the latest windows 10 build

LordOfMice, 2022-11-28:
But "windows 10" is NOT "windows". JFYI - Windows 11 exists and has 22h2 version too. 
And answer to your REAL question:
https://www.overclock.net/threads/usb-mouse-hard-overclocking-2000-hz.1589644/post-29079398

closed 2022-11-28 by LordOfMice