LordOfMice/hidusbf #195: not working in windows 10

TheOldZeref, 2022-12-31:
[image]
I already installed the driver in windows 7/8.1 but when i update to windows 10 it stopped work

TheOldZeref, 2022-12-31:
[image]
I used the folders of hidusbfn and this happen as you can see i have secure boot off

LordOfMice, 2022-12-31:
>as you can see i have secure boot off

readme:
DRIVER folders are intended for REPLACEMENT of
original HIDUSBF drivers and under x64 versions of OS
will run in Test Mode ONLY!

Test Mode On != Secure Boot Off

and JFYI:
https://github.com/LordOfMice/hidusbf/issues/16#issuecomment-1368260189

closed 2022-12-31 by LordOfMice